archlinuxarm/PKGBUILDs #1053: Add XTS support to cubieboard2 kernel.

rookierks, 2015-01-01:
It would be nice if support for XTS could be enabled in linux-sun7i (cubieboard2 kernel), I am particularly interested in support for aes-xts-plain64 as all my external usb disks are encrypted with that.

Currently 'zgrep CRYPTO_XTS /proc/config.gz' returns '# CONFIG_CRYPTO_XTS is not set'. To add XTS support it should be enough to change it to 'CONFIG_CRYPTO_XTS=m',  as in the raspberry pi kernel (or 'CONFIG_CRYPTO_XTS=y' as done for other kernels).

It might also be needed to enable gf128mul, but since that is a dependency of xts it should get enabled when enabling xts.

closed 2015-01-01 by kmihelich

rookierks, 2015-01-02:
Any special reason why this is invalid, such as any know problems with xts and the supported kernel version?

WarheadsSE, 2015-01-02:
https://github.com/archlinuxarm/PKGBUILDs/blob/master/CONTRIBUTING.md