archlinuxarm/PKGBUILDs #1056: systemd: timedatectl crashing on odroid-c1

qjcg, 2015-01-04:
Package: **systemd (218-1)**
Architecture: **armv7**
Device: **odroid-c1**

*What it isn't doing:*
Executing without crashing.

*What it should be doing:*
Executing without crashing.

*Steps to reproduce:*
Run the `timedatectl` command with no arguments.

I get an error like the following every time:

```
# timedatectl
      Local time: Sun 2015-01-04 17:22:51 EST
  Universal time: Sun 2015-01-04 22:22:51 UTC
        RTC time: n/a
       Time zone: America/Montreal (EST, -0500)
     NTP enabled: yes
NTP synchronized: yes
 RTC in local TZ: no
*** Error in `timedatectl': double free or corruption (!prev): 0xb6f14ba0 ***
======= Backtrace: =========
/usr/lib/libc.so.6(+0x6870c)[0xb6d2370c]
/usr/lib/libc.so.6(+0x6e384)[0xb6d29384]
/usr/lib/libc.so.6(+0x6ed2c)[0xb6d29d2c]
======= Memory map: ========
b6b32000-b6cbb000 r--p 00000000 b3:01 530264     /usr/lib/locale/locale-archive
b6cbb000-b6de4000 r-xp 00000000 b3:01 411051     /usr/lib/libc-2.20.so
b6de4000-b6df4000 ---p 00129000 b3:01 411051     /usr/lib/libc-2.20.so
b6df4000-b6df6000 r--p 00129000 b3:01 411051     /usr/lib/libc-2.20.so
b6df6000-b6df7000 rw-p 0012b000 b3:01 411051     /usr/lib/libc-2.20.so
b6df7000-b6dfa000 rw-p 00000000 00:00 0
b6dfa000-b6e0f000 r-xp 00000000 b3:01 410900     /usr/lib/libpthread-2.20.so
b6e0f000-b6e1e000 ---p 00015000 b3:01 410900     /usr/lib/libpthread-2.20.so
b6e1e000-b6e1f000 r--p 00014000 b3:01 410900     /usr/lib/libpthread-2.20.so
b6e1f000-b6e20000 rw-p 00015000 b3:01 410900     /usr/lib/libpthread-2.20.so
b6e20000-b6e22000 rw-p 00000000 00:00 0
b6e22000-b6e3e000 r-xp 00000000 b3:01 411060     /usr/lib/libgcc_s.so.1
b6e3e000-b6e4e000 ---p 0001c000 b3:01 411060     /usr/lib/libgcc_s.so.1
b6e4e000-b6e4f000 rw-p 0001c000 b3:01 411060     /usr/lib/libgcc_s.so.1
b6e4f000-b6e55000 r-xp 00000000 b3:01 410926     /usr/lib/librt-2.20.so
b6e55000-b6e64000 ---p 00006000 b3:01 410926     /usr/lib/librt-2.20.so
b6e64000-b6e65000 r--p 00005000 b3:01 410926     /usr/lib/librt-2.20.so
b6e65000-b6e66000 rw-p 00006000 b3:01 410926     /usr/lib/librt-2.20.so
b6e75000-b6e95000 r-xp 00000000 b3:01 411024     /usr/lib/ld-2.20.so
b6e9e000-b6ea3000 rw-p 00000000 00:00 0
b6ea3000-b6ea4000 r-xp 00000000 00:00 0          [sigpage]
b6ea4000-b6ea5000 r--p 0001f000 b3:01 411024     /usr/lib/ld-2.20.so
b6ea5000-b6ea6000 rw-p 00020000 b3:01 411024     /usr/lib/ld-2.20.so
b6ea6000-b6f0f000 r-xp 00000000 b3:01 393466     /usr/bin/timedatectl
b6f0f000-b6f11000 r--p 00068000 b3:01 393466     /usr/bin/timedatectl
b6f11000-b6f12000 rw-p 0006a000 b3:01 393466     /usr/bin/timedatectl
b6f12000-b6f33000 rw-p 00000000 00:00 0          [heap]
bedc4000-bede5000 rw-p 00000000 00:00 0          [stack]
ffff0000-ffff1000 r-xp 00000000 00:00 0          [vectors]
Aborted (core dumped)
```

rickhull, 2015-02-25:
confirming on cubox-i:

http://www.solid-run.com/community/topic2143.html

and for me, today:

```
[root@cubox ~]# timedatectl
Warning: Ignoring the TZ variable. Reading the system's time zone setting only.

      Local time: Wed 2015-02-25 08:41:02 UTC
  Universal time: Wed 2015-02-25 08:41:02 UTC
        RTC time: Wed 2014-05-28 20:23:56
       Time zone: UTC (UTC, +0000)
     NTP enabled: yes
NTP synchronized: yes
 RTC in local TZ: no
*** Error in `timedatectl': free(): invalid next size (fast): 0x777528d8 ***
======= Backtrace: =========
/usr/lib/libc.so.6(+0x6870c)[0x76d1770c]
/usr/lib/libc.so.6(+0x6e384)[0x76d1d384]
/usr/lib/libc.so.6(+0x6ed2c)[0x76d1dd2c]
======= Memory map: ========
76caf000-76dd8000 r-xp 00000000 b3:01 19944       /usr/lib/libc-2.20.so
76dd8000-76de8000 ---p 00129000 b3:01 19944       /usr/lib/libc-2.20.so
76de8000-76dea000 r--p 00129000 b3:01 19944       /usr/lib/libc-2.20.so
76dea000-76deb000 rw-p 0012b000 b3:01 19944       /usr/lib/libc-2.20.so
76deb000-76dee000 rw-p 00000000 00:00 0 
76dee000-76e03000 r-xp 00000000 b3:01 19919       /usr/lib/libpthread-2.20.so
76e03000-76e12000 ---p 00015000 b3:01 19919       /usr/lib/libpthread-2.20.so
76e12000-76e13000 r--p 00014000 b3:01 19919       /usr/lib/libpthread-2.20.so
76e13000-76e14000 rw-p 00015000 b3:01 19919       /usr/lib/libpthread-2.20.so
76e14000-76e16000 rw-p 00000000 00:00 0 
76e16000-76e32000 r-xp 00000000 b3:01 20179       /usr/lib/libgcc_s.so.1
76e32000-76e42000 ---p 0001c000 b3:01 20179       /usr/lib/libgcc_s.so.1
76e42000-76e43000 rw-p 0001c000 b3:01 20179       /usr/lib/libgcc_s.so.1
76e43000-76e49000 r-xp 00000000 b3:01 19926       /usr/lib/librt-2.20.so
76e49000-76e58000 ---p 00006000 b3:01 19926       /usr/lib/librt-2.20.so
76e58000-76e59000 r--p 00005000 b3:01 19926       /usr/lib/librt-2.20.so
76e59000-76e5a000 rw-p 00006000 b3:01 19926       /usr/lib/librt-2.20.so
76e68000-76e88000 r-xp 00000000 b3:01 19940       /usr/lib/ld-2.20.so
76e91000-76e96000 rw-p 00000000 00:00 0 
76e96000-76e97000 r-xp 00000000 00:00 0           [sigpage]
76e97000-76e98000 r--p 0001f000 b3:01 19940       /usr/lib/ld-2.20.so
76e98000-76e99000 rw-p 00020000 b3:01 19940       /usr/lib/ld-2.20.so
76e99000-76f02000 r-xp 00000000 b3:01 9567        /usr/bin/timedatectl
76f02000-76f04000 r--p 00068000 b3:01 9567        /usr/bin/timedatectl
76f04000-76f05000 rw-p 0006a000 b3:01 9567        /usr/bin/timedatectl
77751000-77772000 rw-p 00000000 00:00 0           [heap]
7e968000-7e989000 rw-p 00000000 00:00 0           [stack]
ffff0000-ffff1000 r-xp 00000000 00:00 0           [vectors]
Aborted (core dumped)
```

pepedog, 2015-02-25:
Could you hwclock -w and try again

rickhull, 2015-02-25:
```
[root@cubox ~]# hwclock -w
[root@cubox ~]# timedatectl
Warning: Ignoring the TZ variable. Reading the system's time zone setting only.

      Local time: Wed 2015-02-25 08:55:46 UTC
  Universal time: Wed 2015-02-25 08:55:46 UTC
        RTC time: Wed 2015-02-25 08:55:46
       Time zone: UTC (UTC, +0000)
     NTP enabled: yes
NTP synchronized: yes
 RTC in local TZ: no
*** Error in `timedatectl': free(): invalid next size (fast): 0x7784a8d8 ***
======= Backtrace: =========
/usr/lib/libc.so.6(+0x6870c)[0x76d4f70c]
/usr/lib/libc.so.6(+0x6e384)[0x76d55384]
/usr/lib/libc.so.6(+0x6ed2c)[0x76d55d2c]
======= Memory map: ========
76ce7000-76e10000 r-xp 00000000 b3:01 19944       /usr/lib/libc-2.20.so
76e10000-76e20000 ---p 00129000 b3:01 19944       /usr/lib/libc-2.20.so
76e20000-76e22000 r--p 00129000 b3:01 19944       /usr/lib/libc-2.20.so
76e22000-76e23000 rw-p 0012b000 b3:01 19944       /usr/lib/libc-2.20.so
76e23000-76e26000 rw-p 00000000 00:00 0 
76e26000-76e3b000 r-xp 00000000 b3:01 19919       /usr/lib/libpthread-2.20.so
76e3b000-76e4a000 ---p 00015000 b3:01 19919       /usr/lib/libpthread-2.20.so
76e4a000-76e4b000 r--p 00014000 b3:01 19919       /usr/lib/libpthread-2.20.so
76e4b000-76e4c000 rw-p 00015000 b3:01 19919       /usr/lib/libpthread-2.20.so
76e4c000-76e4e000 rw-p 00000000 00:00 0 
76e4e000-76e6a000 r-xp 00000000 b3:01 20179       /usr/lib/libgcc_s.so.1
76e6a000-76e7a000 ---p 0001c000 b3:01 20179       /usr/lib/libgcc_s.so.1
76e7a000-76e7b000 rw-p 0001c000 b3:01 20179       /usr/lib/libgcc_s.so.1
76e7b000-76e81000 r-xp 00000000 b3:01 19926       /usr/lib/librt-2.20.so
76e81000-76e90000 ---p 00006000 b3:01 19926       /usr/lib/librt-2.20.so
76e90000-76e91000 r--p 00005000 b3:01 19926       /usr/lib/librt-2.20.so
76e91000-76e92000 rw-p 00006000 b3:01 19926       /usr/lib/librt-2.20.so
76ea0000-76ec0000 r-xp 00000000 b3:01 19940       /usr/lib/ld-2.20.so
76ec9000-76ece000 rw-p 00000000 00:00 0 
76ece000-76ecf000 r-xp 00000000 00:00 0           [sigpage]
76ecf000-76ed0000 r--p 0001f000 b3:01 19940       /usr/lib/ld-2.20.so
76ed0000-76ed1000 rw-p 00020000 b3:01 19940       /usr/lib/ld-2.20.so
76ed1000-76f3a000 r-xp 00000000 b3:01 9567        /usr/bin/timedatectl
76f3a000-76f3c000 r--p 00068000 b3:01 9567        /usr/bin/timedatectl
76f3c000-76f3d000 rw-p 0006a000 b3:01 9567        /usr/bin/timedatectl
77849000-7786a000 rw-p 00000000 00:00 0           [heap]
7eb5c000-7eb7d000 rw-p 00000000 00:00 0           [stack]
ffff0000-ffff1000 r-xp 00000000 00:00 0           [vectors]
Aborted (core dumped)
```

michaellass, 2015-03-02:
I had the same issue on the raspberry pi 2 but it disappeared after setting a timezone, e.g. by calling 

    timedatectl set-timezone Europe/Berlin

I guess this is an upstream bug but I didn't have a closer look yet.

michaellass, 2015-03-02:
I traced the issue down and it is already fixed upstream:
http://cgit.freedesktop.org/systemd/systemd/commit/?id=e17fb3c1dd21929119968325c46ab98630446032
That change is included in systemd 219 which is already in Arch's [testing] repository.

kmihelich, 2015-04-20:
Fixed in systemd 219-5.

closed 2015-04-20 by kmihelich