archlinuxarm/PKGBUILDs #1060: linux-imx6-cubox-dt-3.14.27-1-armv7h: error on loading bridge-module

ineiti, 2015-01-10:
Package: linux-imx6-cubox-dt-3.14.27-1-armv7h
Architecture: armv7h
Device: Cubox I4P-300D and Cubox I

What it isn't doing:
loading bridge-module

What it should be doing:
load bridge-module

Steps to reproduce:
```modprobe bridge```

Works fine in:
linux-imx6-cubox-dt-3.14.25-1-armv7h

Error-message:
````
[   34.249268] Unable to handle kernel paging request at virtual address 7f20a2f0
[   34.256832] pgd = dcef8000
[   34.259695] [7f20a2f0] *pgd=6cd52811, *pte=6c7d54df, *ppte=6c7d565e
[   34.266430] Internal error: Oops: 80f [#1] SMP ARM
[   34.271379] Modules linked in: bridge(+) ipt_MASQUERADE iptable_nat nf_conntrack_ipv4 nf_defrag_ipv4 nf_nat_ipv4 nf_nat nf_conntrack brcmfmac brcmutil ir_lirc_codec cfg80211 rfkill lirc_dev ir_jvc_decoder ir_sony_decoder ir_sanyo_decoder ir_mce_kbd_decoder caam ir_rc6_decoder ir_rc5_decoder ir_nec_decoder gpio_ir_recv sch_fq_codel
[   34.302272] CPU: 2 PID: 572 Comm: modprobe Not tainted 3.14.27-1-ARCH #1
[   34.309185] task: dc786e00 ti: dcec2000 task.ti: dcec2000
[   34.314791] PC is at patch_text+0xc/0x1c
[   34.318850] LR is at __jump_label_update+0x64/0x6c
[   34.323801] pc : [<807467a8>]    lr : [<800bb128>]    psr: 800c0013
[   34.323801] sp : dcec3db0  ip : 000174e4  fp : 00000000
[   34.335631] r10: 7f2127dc  r9 : dcaa50c8  r8 : 00000000
[   34.341021] r7 : 00000001  r6 : 80b54acc  r5 : 7f212794  r4 : 7f2126f8
[   34.347748] r3 : 7f20a2f0  r2 : 80a86534  r1 : ea000007  r0 : 7f20a2f0
[   34.354493] Flags: Nzcv  IRQs on  FIQs on  Mode SVC_32  ISA ARM  Segment user
[   34.361846] Control: 10c53c7d  Table: 6cef804a  DAC: 00000015
[   34.367770] Process modprobe (pid: 572, stack limit = 0xdcec2238)
[   34.374051] Stack: (0xdcec3db0 to 0xdcec4000)
[   34.378554] 3da0:                                     00000001 dcaa5240 80b54acc 00000001
[   34.386985] 3dc0: 00000000 800bb17c 80b54acc 00000001 00000007 dcaa50c0 7f212600 800bb29c
[   34.395415] 3de0: 000002a0 7f212600 7f212620 806588dc 00000000 806589b8 7f212914 7f21a000
[   34.403845] 3e00: 7f2127a0 dcaa50c0 00000001 7f21a1a4 7f212914 806246f4 00000000 00000000
[   34.412272] 3e20: 7f21a000 7f21a050 dcec2030 8000881c 00000000 800dcf60 7f21b000 80a940a8
[   34.420699] 3e40: 80a940a8 80b38900 00000000 800dcf60 7f222000 80a940a8 80a940a8 80b38900
[   34.429126] 3e60: 7f221fff 00000000 dcaa50c8 7f21b000 00000000 7f222000 dcaa50c0 00000001
[   34.437554] 3e80: dcaa50c8 dcec3f58 00000001 7f2127a0 dcaa50c0 00000001 dcaa50c8 7f2127dc
[   34.445981] 3ea0: 7f212794 8008b298 7f2127a0 00007fff 800881f8 00000000 dcaecb00 00000000
[   34.454407] 3ec0: dcec3f58 7f2127a0 dcec2028 f08c2ef8 00000000 dcec3f60 000003ca dcec3f8c
[   34.462834] 3ee0: dcb10040 000000d2 000002d2 8000e4c4 dcec2000 7f210024 00000001 00000000
[   34.471259] 3f00: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[   34.479686] 3f20: 00000000 00000000 00000000 00000000 76872008 00020f0b 76872008 00023bf8
[   34.488112] 3f40: 00000080 8000e4c4 dcec2000 00000200 013a0300 8008b8b8 f08a2000 00020f0b
[   34.496538] 3f60: f08b6900 f08b66fd f08bfe70 00012000 00014d20 00000000 00000000 00000000
[   34.504963] 3f80: 0000002f 00000030 00000025 00000029 00000016 00000000 00000000 76872008
[   34.513391] 3fa0: 00040000 8000e4a8 00000000 76872008 76872008 00020f0b 00023bf8 2a734300
[   34.521820] 3fc0: 00000000 76872008 00040000 00000080 00023bf8 00000000 013a01e0 013a0300
[   34.530248] 3fe0: 76eab430 7ec3d8c8 00018994 76eab440 600c0010 76872008 7f77e821 7f77ec21
[   34.538694] [<807467a8>] (patch_text) from [<800bb128>] (__jump_label_update+0x64/0x6c)
[   34.554220] [<800bb128>] (__jump_label_update) from [<800bb17c>] (jump_label_update+0x4c/0x90)
[   34.570446] [<800bb17c>] (jump_label_update) from [<800bb29c>] (static_key_slow_inc+0xdc/0xe0)
[   34.586751] [<800bb29c>] (static_key_slow_inc) from [<806588dc>] (nf_register_hook+0xbc/0xc8)
[   34.602834] [<806588dc>] (nf_register_hook) from [<806589b8>] (nf_register_hooks+0x38/0x78)
[   34.618688] [<806589b8>] (nf_register_hooks) from [<7f21a1a4>] (br_netfilter_init+0x38/0xe94 [bridge])
[   34.635470] [<7f21a1a4>] (br_netfilter_init [bridge]) from [<7f21a050>] (br_init+0x50/0xd4 [bridge])
[   34.652069] [<7f21a050>] (br_init [bridge]) from [<8000881c>] (do_one_initcall+0xf8/0x154)
[   34.667768] [<8000881c>] (do_one_initcall) from [<8008b298>] (load_module+0x1b58/0x20d0)
[   34.683308] [<8008b298>] (load_module) from [<8008b8b8>] (SyS_init_module+0xa8/0xec)
[   34.698302] [<8008b8b8>] (SyS_init_module) from [<8000e4a8>] (__sys_trace_return+0x0/0x18)
[   34.713622] Code: e12fff12 e3062534 e1a03000 e34820a8 (e4831004)
[   34.726696] ---[ end trace 23a46f1311ed527a ]---
```

pepedog, 2015-01-10:
I think this is CONFIG_JUMP_LABEL
will test and fix

closed 2015-01-10 by pepedog

ineiti, 2015-01-13:
Le 11/01/2015 00:12, pepedog a écrit :
> Closed #1060 <https://github.com/archlinuxarm/PKGBUILDs/issues/1060> via
> 4e08a54
> <https://github.com/archlinuxarm/PKGBUILDs/commit/4e08a5446963960dcf78466b617063825956900b>.

Works now - thank you very much

commit 4e08a544 (2015-01-10): core/linux-imx6-cubox-dt remove CONFIG_JUMP_LABEL fixes #1060