archlinuxarm/PKGBUILDs #1062: ghc 7.8.2: built executables fail with 'schedule: re-entered unsafely' on Odroid U3

alexeicolin, 2015-01-11:
**Package**: ghc 7.8.2-3 (related llvm 3.5.0-3)
**Architecture**: armv7h
**Device**: Hardkernel Odroid U3

What it isn't doing: A hello-world (and other) executables built with ghc, when run, print the error message pasted below and return a non-zero exit code.

What it should be doing: A hello-world executable compiled with ghc should run correctly and return a zero exit code.

**Steps to reproduce**:

    $ cat hello.hs
    main = putStrLn "hello"
    $ ghc -o hello hello.hs
    [1 of 1] Compiling Main             ( hello.hs, hello.o )
    You are using a new version of LLVM that hasn't been tested yet!
    We will try though...
    Linking hello ...
    $ ./hello
    hello: schedule: re-entered unsafely.
       Perhaps a 'foreign import unsafe' should be 'safe'?
    $ echo $?
    1

When built with `-threaded`, `hello` does not produce the same error, but it never returns.

[Link to verbose ghc compile log.](https://gist.github.com/alexeicolin/6873160b9ed02ec0a1b9).
The log shows args to `llc`: `-mattr=+v7,+vfp3,+d16`, in case this is related to #175.

**More details**:

This problem also manifests when trying to build GHC 7.8.4-1 package from c20d6625: an intermediate executable `ghc-pwd` fails in the above-described way:

    configure: Building in-tree ghc-pwd
    You are using a new version of LLVM that hasn't been tested yet!
    We will try though...
    checking for path to top of build tree... ghc-pwd: schedule: re-entered unsafely.
       Perhaps a 'foreign import unsafe' should be 'safe'?
    configure: error: cannot determine current directory
    ==> ERROR: A failure occurred in build().

Since it complains about LLVM being too new, I tried building an older one (v3.4-2 and v3.4.1-1 based on dates before the ghc 7.8.2 commit) as well as rebuilding the latest llvm 3.5.0-3 from Arch repo (which I have installed and is being used in the ghc build). All fail to build due to gcc getting stuck on compiling `Function.cpp` from the IR library (deterministically). The gcc command that never finishes (after outputting what looks like completed object code) is the following. When this command is rerun manually outside of makepkg, same problem.

    acolin     407 98.0 13.9 305560 288776 pts/9   R+   17:14  13:23 /usr/lib/gcc/armv7l-unknown-linux-gnueabihf/4.9.2/cc1plus -quiet -I /home/acolin/dev/packages/llvm/llvm/trunk/src/llvm-3.5.0.src/include -I /home/acolin/dev/packages/llvm/llvm/trunk/src/llvm-3.5.0.src/lib/IR -I /usr/lib/libffi-3.2.1/include -I /usr/lib/libffi-3.2.1/include -I /usr/lib/libffi-3.2.1/include -MMD /home/acolin/dev/packages/llvm/llvm/trunk/src/llvm-3.5.0.src/lib/IR/Release/Function.d -MF /home/acolin/dev/packages/llvm/llvm/trunk/src/llvm-3.5.0.src/lib/IR/Release/Function.d.tmp -MP -MT /home/acolin/dev/packages/llvm/llvm/trunk/src/llvm-3.5.0.src/lib/IR/Release/Function.o -MT /home/acolin/dev/packages/llvm/llvm/trunk/src/llvm-3.5.0.src/lib/IR/Release/Function.d -D_GNU_SOURCE -D NDEBUG -D _GNU_SOURCE -D __STDC_CONSTANT_MACROS -D __STDC_FORMAT_MACROS -D __STDC_LIMIT_MACROS -D _FORTIFY_SOURCE=2 -D _FORTIFY_SOURCE=2 -D _FORTIFY_SOURCE=2 Function.cpp -quiet -dumpbase Function.cpp -march=armv7-a -mfloat-abi=hard -mfpu=vfpv3-d16 -march=armv7-a -mfloat-abi=hard -mfpu=vfpv3-d16 -march=armv7-a -mfloat-abi=hard -mfpu=vfpv3-d16 -mtls-dialect=gnu -auxbase-strip /home/acolin/dev/packages/llvm/llvm/trunk/src/llvm-3.5.0.src/lib/IR/Release/Function.o -O3 -O2 -O2 -O2 -Woverloaded-virtual -Wcast-qual -Wpedantic -Wno-long-long -Wall -Wextra -Wno-unused-parameter -Wwrite-strings -Wno-maybe-uninitialized -Wno-missing-field-initializers -std=c++11 -fomit-frame-pointer -fvisibility-inlines-hidden -fno-exceptions -fPIC -ffunction-sections -fdata-sections -fstack-protector --param ssp-buffer-size=4 --param ssp-buffer-size=4 --param ssp-buffer-size=4 -o -

kmihelich, 2015-01-11:
ghc is very broken on ARM.  The packages only remain to potentially help building in the future.

closed 2015-01-11 by kmihelich

alexeicolin, 2015-01-12:
I gathered that much. The point of this report was to track the fact that ghc does not work. Yet, you just close the issue. Why not leave it open until it becomes invalid or is fixed one way or another? The next person who finds ghc not working would quickly find the report and not waste time scratching their head.

Your policy is baffling. So far I've submitted a PR that was closed without a concrete reason and a bug report that was closed for no good reason. :'(

alexeicolin, 2015-01-12:
I would either keep this open or keep the GHC package out of master branch for as long as it does not work at all. Development branches are a better place for things that might "potentially help building in the future". If a package is in master,  and there are no open issues, then the user can't do better than assume that the package has been working in the past and should be more-or-less functional. You disagree?