archlinuxarm/PKGBUILDs #1072: uboot-sunxi 2015-1 ignores uEnv.txt

somenxavier, 2015-01-23:
[this](https://github.com/archlinuxarm/PKGBUILDs/commit/92b2d45117990d2c4ea7022c83549519b143b705) ignores the file uEnv.txt. In my case:

```
# cat uEnv.txt
root=/dev/sda1 rw rootwait
mmcroot=UUID=81e9a6bd-140a-43ff-8409-8623a5d42adb rw rootwait
mmcrootfstype=ext4
optargs=elevator=noop ethaddr=00:CE:39:B7:74:4E rootfstype=ext4
```

If you close this ticket, please show how can I set the kernel parameters with the new uboot-sunxi package. Because previously it worked perfectly.

closed 2015-01-23 by kmihelich

somenxavier, 2015-01-23:
But there is an issue here: before all was fine and now uEnv.txt is ignored.Do not scant words !

somenxavier, 2015-01-23:
I use Olimex A20 Micro

WarheadsSE, 2015-01-23:
FORUM.