archlinuxarm/PKGBUILDs #1074: dkms-mt7601 no longer compiles as of 3.18.3-1

VonL, 2015-01-26:
Tested on a raspberry pi, kernel version 3.18.3-1-ARCH. DKMS log is as follows:

    DKMS make.log for mt7601-v3.0.0.4 for kernel 3.18.3-1-ARCH (armv6l)
Mon Jan 26 10:12:11 CET 2015
make -C tools
make[1]: Entering directory '/var/lib/dkms/mt7601/v3.0.0.4/build/tools'
gcc -g bin2h.c -o bin2h
make[1]: Leaving directory '/var/lib/dkms/mt7601/v3.0.0.4/build/tools'
/var/lib/dkms/mt7601/v3.0.0.4/build/tools/bin2h
cp -f os/linux/Makefile.6 /var/lib/dkms/mt7601/v3.0.0.4/build/os/linux/Makefile
make -C /lib/modules/3.18.3-1-ARCH/build SUBDIRS=/var/lib/dkms/mt7601/v3.0.0.4/build/os/linux modules
make[1]: Entering directory '/usr/lib/modules/3.18.3-1-ARCH/build'
  CC [M]  /var/lib/dkms/mt7601/v3.0.0.4/build/os/linux/../../os/linux/rt_profile.o
/var/lib/dkms/mt7601/v3.0.0.4/build/os/linux/../../os/linux/rt_profile.c: In function 'announce_802_3_packet':
/var/lib/dkms/mt7601/v3.0.0.4/build/os/linux/../../os/linux/rt_profile.c:331:16: warning: unused variable 'pAd' [-Wunused-variable]
  RTMP_ADAPTER *pAd = (RTMP_ADAPTER *)pAdSrc;
                ^
  CC [M]  /var/lib/dkms/mt7601/v3.0.0.4/build/os/linux/../../sta/assoc.o
  CC [M]  /var/lib/dkms/mt7601/v3.0.0.4/build/os/linux/../../sta/auth.o
  CC [M]  /var/lib/dkms/mt7601/v3.0.0.4/build/os/linux/../../sta/auth_rsp.o
  CC [M]  /var/lib/dkms/mt7601/v3.0.0.4/build/os/linux/../../sta/sync.o
/var/lib/dkms/mt7601/v3.0.0.4/build/os/linux/../../sta/sync.c: In function 'PeerBeacon':
/var/lib/dkms/mt7601/v3.0.0.4/build/os/linux/../../sta/sync.c:2180:12: warning: passing argument 8 of 'StaAddMacTableEntry' from incompatible pointer type
            ie_list,
            ^
In file included from /var/lib/dkms/mt7601/v3.0.0.4/build/include/rt_config.h:59:0,
                 from /var/lib/dkms/mt7601/v3.0.0.4/build/os/linux/../../sta/sync.c:28:
/var/lib/dkms/mt7601/v3.0.0.4/build/include/rtmp.h:7892:9: note: expected 'struct IE_LISTS *' but argument is of type 'struct BCN_IE_LIST *'
 BOOLEAN StaAddMacTableEntry(
         ^
  CC [M]  /var/lib/dkms/mt7601/v3.0.0.4/build/os/linux/../../sta/sanity.o
  CC [M]  /var/lib/dkms/mt7601/v3.0.0.4/build/os/linux/../../sta/rtmp_data.o
/var/lib/dkms/mt7601/v3.0.0.4/build/os/linux/../../sta/rtmp_data.c: In function 'STAHandleRxDataFrame':
/var/lib/dkms/mt7601/v3.0.0.4/build/os/linux/../../sta/rtmp_data.c:523:33: warning: passing argument 2 of 'MacTableLookup' from incompatible pointer type
    pEntry = MacTableLookup(pAd, &pHeader->Addr2);
                                 ^
In file included from /var/lib/dkms/mt7601/v3.0.0.4/build/include/rt_config.h:59:0,
                 from /var/lib/dkms/mt7601/v3.0.0.4/build/os/linux/../../sta/rtmp_data.c:28:
/var/lib/dkms/mt7601/v3.0.0.4/build/include/rtmp.h:8429:18: note: expected 'UCHAR *' but argument is of type 'UCHAR (*)[6]'
 MAC_TABLE_ENTRY *MacTableLookup(RTMP_ADAPTER *pAd, UCHAR *pAddr);
                  ^
  CC [M]  /var/lib/dkms/mt7601/v3.0.0.4/build/os/linux/../../sta/connect.o
  CC [M]  /var/lib/dkms/mt7601/v3.0.0.4/build/os/linux/../../sta/wpa.o
  CC [M]  /var/lib/dkms/mt7601/v3.0.0.4/build/os/linux/../../sta/sta_cfg.o
/var/lib/dkms/mt7601/v3.0.0.4/build/os/linux/../../sta/sta_cfg.c: In function 'RTMPIoctlRF':
/var/lib/dkms/mt7601/v3.0.0.4/build/os/linux/../../sta/sta_cfg.c:5306:7: warning: format '%X' expects argument of type 'unsigned int', but argument 5 has type 'LONG' [-Wformat=]
       sprintf(msg+strlen(msg), "BANK%d_R%02d:%02X  ", bank_Id, rfId, rfValue);
       ^
/var/lib/dkms/mt7601/v3.0.0.4/build/os/linux/../../sta/sta_cfg.c:5359:27: warning: passing argument 2 of 'RtmpDrvAllRFPrint' from incompatible pointer type
   RtmpDrvAllRFPrint(NULL, msg, strlen(msg));
                           ^
In file included from /var/lib/dkms/mt7601/v3.0.0.4/build/include/rt_config.h:64:0,
                 from /var/lib/dkms/mt7601/v3.0.0.4/build/os/linux/../../sta/sta_cfg.c:28:
/var/lib/dkms/mt7601/v3.0.0.4/build/include/rt_os_util.h:668:6: note: expected 'UINT32 *' but argument is of type 'PSTRING'
 VOID RtmpDrvAllRFPrint(
      ^
/var/lib/dkms/mt7601/v3.0.0.4/build/os/linux/../../sta/sta_cfg.c:5209:22: warning: unused variable 'rf_bank' [-Wunused-variable]
  UCHAR    regRF = 0, rf_bank = 0;
                      ^
/var/lib/dkms/mt7601/v3.0.0.4/build/os/linux/../../sta/sta_cfg.c: In function 'RTMPIoctlShow':
/var/lib/dkms/mt7601/v3.0.0.4/build/os/linux/../../sta/sta_cfg.c:5766:85: error: macro "__DATE__" might prevent reproducible builds [-Werror=date-time]
             snprintf(extra, size, "Driver version-%s, %s %s\n", STA_DRIVER_VERSION, __DATE__, __TIME__ );
                                                                                     ^
/var/lib/dkms/mt7601/v3.0.0.4/build/os/linux/../../sta/sta_cfg.c:5766:95: error: macro "__TIME__" might prevent reproducible builds [-Werror=date-time]
             snprintf(extra, size, "Driver version-%s, %s %s\n", STA_DRIVER_VERSION, __DATE__, __TIME__ );
                                                                                               ^
/var/lib/dkms/mt7601/v3.0.0.4/build/os/linux/../../sta/sta_cfg.c: In function 'RtmpIoctl_rt_ioctl_siwgenie':
/var/lib/dkms/mt7601/v3.0.0.4/build/os/linux/../../sta/sta_cfg.c:7610:13: warning: assignment from incompatible pointer type
     eid_ptr = pAd->StaCfg.pWpaAssocIe;
             ^
cc1: some warnings being treated as errors
scripts/Makefile.build:257: recipe for target '/var/lib/dkms/mt7601/v3.0.0.4/build/os/linux/../../sta/sta_cfg.o' failed
make[2]: *** [/var/lib/dkms/mt7601/v3.0.0.4/build/os/linux/../../sta/sta_cfg.o] Error 1
Makefile:1381: recipe for target '_module_/var/lib/dkms/mt7601/v3.0.0.4/build/os/linux' failed
make[1]: *** [_module_/var/lib/dkms/mt7601/v3.0.0.4/build/os/linux] Error 2
make[1]: Leaving directory '/usr/lib/modules/3.18.3-1-ARCH/build'
Makefile:394: recipe for target 'LINUX' failed
make: *** [LINUX] Error 2

VonL, 2015-01-26:
The following change to file sta/sta_cfg.c is enough of a workaround to let the module build:
    snprintf(extra, size, "Driver version-%s, %s %s\n", STA_DRIVER_VERSION, __DATE__, __TIME__ );
to
    snprintf(extra, size, "Driver version-%s, (date and time removed for sanity)\n", STA_DRIVER_VERSION );

Does anyone know where I should turn to "publish" these changes? I'd like to make sure other people do not encounter this problem.

closed 2015-01-26 by kmihelich

cbonnissent, 2015-02-09:
Thanks for the advice !