archlinuxarm/PKGBUILDs #1079: ntop: gets killed with FPE

ineiti, 2015-01-31:
Package: community/ntop 5.0.1-6
Architecture: armv5
Device: DreamPlug

What it isn't doing:
Stop quitting with a FPE

What it should be doing:
Nicely output network traffic

Steps to reproduce:
Install and run, for example with

```ntop -i ppp0````

and see it quitting with

```Floating point exception```

pepedog, 2015-01-31:
It might be the older kernel you might be running, it works fine on linux-kirkwood

kmihelich, 2015-01-31:
Works fine here too, can't reproduce.

closed 2015-01-31 by kmihelich

ineiti, 2015-01-31:
Le 31/01/2015 17:16, Kevin Mihelich a écrit :
> Works fine here too, can't reproduce.

Are you using -dt kernel? I'm on 3.18.1-1, but not -dt. Just updated 
everything, but still not working.