archlinuxarm/PKGBUILDs #1082: sshfs segfault

nwf, 2015-02-04:
sshfs as packaged crashes on my Pogoplug v4, running the packaged ALARM kernel.  uname is
```
Linux nwfdmz 3.18.5-1-ARCH #1 PREEMPT Sat Jan 31 02:35:20 MST 2015 armv5tel GNU/Linux
```

Rebuilding sshfs with debugging symbols from the `PKGBUILD` from https://projects.archlinux.org/svntogit/community.git/plain/trunk/PKGBUILD?h=packages/sshfs and `CFLAGS=-g` and running inside `gdb` shows:

```
Reading symbols from /home/aur/sshfs/src/sshfs-fuse-2.5/sshfs...done.
(gdb) run
Starting program: /home/aur/sshfs/src/sshfs-fuse-2.5/sshfs -d -p 2222 -o IdentityFile=/root/sftp-key -o max_write=1024 sftp-nwfdmz@172.29.2.1:/ /mnt/sftp
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/usr/lib/libthread_db.so.1".
FUSE library version: 2.9.3
nullpath_ok: 0
nopath: 0
utime_omit_ok: 0
[New Thread 0xb6c5f460 (LWP 8408)]
[New Thread 0xb62ff460 (LWP 8409)]
unique: 1, opcode: INIT (26), nodeid: 0, insize: 56, pid: 0
INIT: 7.23
flags=0x0003f7fb
max_readahead=0x00020000
[New Thread 0xb5aff460 (LWP 8410)]
   INIT: 7.19
   flags=0x00000011
   max_readahead=0x00020000
   max_write=0x00020000
   max_background=0
   congestion_threshold=0
   unique: 1, success, outsize: 40
unique: 2, opcode: GETATTR (3), nodeid: 1, insize: 56, pid: 7047
getattr /
   unique: 2, success, outsize: 120
unique: 3, opcode: OPENDIR (27), nodeid: 1, insize: 48, pid: 7047
   unique: 3, success, outsize: 32
unique: 4, opcode: READDIR (28), nodeid: 1, insize: 80, pid: 7047
getdir[0]
   unique: 4, success, outsize: 112
unique: 5, opcode: READDIR (28), nodeid: 1, insize: 80, pid: 7047
   unique: 5, success, outsize: 16
unique: 6, opcode: RELEASEDIR (29), nodeid: 1, insize: 64, pid: 0
   unique: 6, success, outsize: 16
[New Thread 0xb50ff460 (LWP 8412)]
unique: 7, opcode: GETATTR (3), nodeid: 1, insize: 56, pid: 7047
getattr /
   unique: 7, success, outsize: 120
unique: 8, opcode: LOOKUP (1), nodeid: 1, insize: 48, pid: 7047
LOOKUP /backups
getattr /backups
   NODEID: 2
   unique: 8, success, outsize: 144
unique: 9, opcode: LOOKUP (1), nodeid: 1, insize: 48, pid: 8413                                                                                                                                            [33/94]
LOOKUP /backups
getattr /backups
   NODEID: 2
   unique: 9, success, outsize: 144
unique: 10, opcode: LOOKUP (1), nodeid: 2, insize: 45, pid: 8413
LOOKUP /backups/test
getattr /backups/test
   unique: 10, error: -2 (No such file or directory), outsize: 16
unique: 11, opcode: CREATE (35), nodeid: 2, insize: 61, pid: 8413
create flags: 0x20841 /backups/test 0100644 umask=0022
   create[3037725944] flags: 0x20841 /backups/test
fgetattr[3037725944] /backups/test
   NODEID: 3
   unique: 11, success, outsize: 160
unique: 12, opcode: FLUSH (25), nodeid: 3, insize: 64, pid: 8413
flush[0]

Program received signal SIGSEGV, Segmentation fault.
[Switching to Thread 0xb6c5f460 (LWP 8408)]
0x00017ef4 in sshfs_file_is_conn (sf=0x0) at sshfs.c:2414
2414            ret = (sf->connver == sshfs.connver);
(gdb) bt
#0  0x00017ef4 in sshfs_file_is_conn (sf=0x0) at sshfs.c:2414
#1  0x00018358 in sshfs_flush (path=0xb6300628 "/backups/test", fi=0xb6c5ece8) at sshfs.c:2518
#2  0xb6fafd6c in fuse_fs_flush () from /usr/lib/libfuse.so.2
Backtrace stopped: Cannot access memory at address 0x288e4
(gdb) up
#1  0x00018358 in sshfs_flush (path=0xb6300628 "/backups/test", fi=0xb6c5ece8) at sshfs.c:2518
2518            if (!sshfs_file_is_conn(sf))
(gdb) print *fi
$1 = {flags = 0, fh_old = 0, writepage = 0, direct_io = 0, keep_cache = 0, flush = 1, nonseekable = 0, flock_release = 0, padding = 0, fh = 0, lock_owner = 10752985285259444149}
```

I'm insufficiently versed in FUSE to know exactly what's going on beyond saying that it's crashing due to dereferencing the "fh = 0" value.

What more can I tell you to be useful?  What should I try?

closed 2015-02-04 by kmihelich

nwf, 2015-02-05:
I updated my comment to follow the template; is it still "Invalid"?

kmihelich, 2015-02-05:
Alignment issue, fixed with rebuild of sshfs 2.5-1.1.  Update your system and re-mount any sshfs mounts.

nwf, 2015-02-05:
I hate to pester you about this, but could you explain what changes you made to the build to fix the alignment issue?  If a rebuild alone sufficed I would have thought that my rebuild to add "CFLAGS=-g" would also have fixed it.

I've not been able to find the archlinuxarm community PKGBUILD for sshfs so I was assuming that you were pulling from the upstream arch one; is this not true?

Thanks much.

nwf, 2015-02-05:
Sigh.  While things definitely improved with 2.5-1.1, they are not perfect.  Running an identical sftp command as above and doing "BUP_DIR=/mnt/sftp/backups bup init" causes another similar crash:

```
...
create flags: 0x200c1 /backups/hooks/pre-commit.sample 0100755 umask=0022
   create[271344] flags: 0x200c1 /backups/hooks/pre-commit.sample
fgetattr[271344] /backups/hooks/pre-commit.sample
   NODEID: 8
   unique: 16, success, outsize: 160
unique: 17, opcode: GETXATTR (22), nodeid: 8, insize: 68, pid: 12344
   unique: 17, error: -38 (Function not implemented), outsize: 16
unique: 18, opcode: WRITE (16), nodeid: 8, insize: 1722, pid: 12344
write[0] 1642 bytes to 0 flags: 0x20001
Cannot access memory at address 0x0

Program received signal SIGSEGV, Segmentation fault.
[Switching to Thread 0xb62ff460 (LWP 12339)]
Cannot access memory at address 0x0
0x00016a7c in ?? ()
```

As I cannot recreate 2.5-1.1 locally, I am afraid I cannot offer debugging symbols or meaningful backtraces or any such for this crash.