archlinuxarm/PKGBUILDs #1086: linux-raspberrypi: 3.18.5-3 kernel panic with a keyboard

Lartza, 2015-02-07:
Package: linux-raspberrypi
Architecture: armv7h
Device: Raspberry Pi 2 Model B

What it isn't doing: Booting with a keyboard plugged in

What it should be doing: Booting with a keyboard plugged in

Steps to reproduce:
Install a fresh system with
ArchLinuxARM-2015.02-rpi-2-rootfs.tar.gz           07-Feb-2015 07:39
or upgrade linux-raspberrypi of the earlier rootfs release, boot with a keyboard plugged in (and possibly press a button)
System will freeze, or output a trace for the kernel panic

pepedog, 2015-02-07:
I can confirm, touch a key and ...
I don't have time to look into this. Not sure if memory split also affects kodi too

kmihelich, 2015-02-07:
They've managed to produce the first device that kills itself trying address high memory, a round of applause please.

Temporary fix: set gpu_mem=256

kmihelich, 2015-02-07:
Will be fixed in 3.18.6-1, which will be in the repos soon.  Highmem support is broken so you're only going to get a maximum of 740MB usable RAM, regardless of setting gpu_mem < 256.

closed 2015-02-07 by kmihelich

morganm, 2015-02-21:
The [forum thread](http://archlinuxarm.org/forum/viewtopic.php?f=64&t=8387) says to ask here:

Is there any way to fix this issue so that the full 1G is accessible and not just 741M? I have two pi2's, one with Arch and one with Raspbian. Raspbian can see the full 1G, but due to this issue, Arch cannot. 

```
# Raspbian
> grep '^NAME' /etc/os-release
NAME="Raspbian GNU/Linux"
> uname -s -r -m
Linux 3.18.7-v7+ armv7l
> head -1 /proc/meminfo
MemTotal:         949328 kB
```

```
# Arch
> grep '^NAME' /etc/os-release
NAME="Arch Linux ARM"
> uname -s -r -m
Linux 3.18.7-6-ARCH armv7l
> head -1 /proc/meminfo
MemTotal:         759644 kB
```

I haven't been able to find any 'keyboard of death' threads for Raspbian, so I assume they're doing something different to get the full 1G without being subject to this bug?

Lartza, 2015-02-21:
config.v7 lines 428 and 429
```
CONFIG_VMSPLIT_3G=y
# CONFIG_VMSPLIT_2G is not set
```
change to 
```
# CONFIG_VMSPLIT_3G is not set
CONFIG_VMSPLIT_2G=y
```
and build the kernel yourself, for now