archlinuxarm/PKGBUILDs #1090: Perl module Aspell installation

oserin, 2015-02-09:
Hi I am pretty new to this, can you please tell me whatelse is needed ?

[root@alarmpi ~]# cpanm Text::Aspell
--> Working on Text::Aspell
Fetching http://www.cpan.org/authors/id/H/HA/HANK/Text-Aspell-0.09.tar.gz ... OK
Configuring Text-Aspell-0.09 ... OK
Building and testing Text-Aspell-0.09 ... FAIL
! Installing Text::Aspell failed. See /root/.cpanm/work/1423512491.12855/build.log for details. Retry with --force to force install it.
[root@alarmpi ~]# cat /root/.cpanm/work/1423512491.12855/build.log 
cpanm (App::cpanminus) 1.7025 on perl 5.020001 built for armv6l-linux-thread-multi
Work directory is /root/.cpanm/work/1423512491.12855
You have make /usr/bin/make
You have /usr/bin/wget
You have /usr/bin/tar: tar (GNU tar) 1.28
Copyright (C) 2014 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>.
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.

Written by John Gilmore and Jay Fenlason.
You have /usr/bin/unzip
Searching Text::Aspell on cpanmetadb ...
--> Working on Text::Aspell
Fetching http://www.cpan.org/authors/id/H/HA/HANK/Text-Aspell-0.09.tar.gz
-> OK
Unpacking Text-Aspell-0.09.tar.gz
Entering Text-Aspell-0.09
Checking configure dependencies from META.yml
Configuring Text-Aspell-0.09
Running Makefile.PL
Checking if your kit is complete...
Looks good
Generating a Unix-style Makefile
Writing Makefile for Text::Aspell
Writing MYMETA.yml and MYMETA.json
-> OK
Checking dependencies from MYMETA.json ...
Checking if you have ExtUtils::MakeMaker 0 ... Yes (6.98)
Checking if you have Test::More 0 ... Yes (1.001002)
Building and testing Text-Aspell-0.09
cp Aspell.pm blib/lib/Text/Aspell.pm
Running Mkbootstrap for Text::Aspell ()
chmod 644 Aspell.bs
/usr/bin/perl /usr/share/perl5/core_perl/ExtUtils/xsubpp -noprototypes -typemap /usr/share/perl5/core_perl/ExtUtils/typemap -typemap typemap  Aspell.xs > Aspell.xsc && mv Aspell.xsc Aspell.c
cc -c   -D_REENTRANT -D_GNU_SOURCE -fwrapv -fno-strict-aliasing -pipe -fstack-protector -I/usr/local/include -D_LARGEFILE_SOURCE -D_FILE_OFFSET_BITS=64 -march=armv6 -mfloat-abi=hard -mfpu=vfp -O2 -pipe -fstack-protector --param=ssp-buffer-size=4   -DVERSION=\"0.09\" -DXS_VERSION=\"0.09\" -fPIC "-I/usr/lib/perl5/core_perl/CORE"   Aspell.c
rm -f blib/arch/auto/Text/Aspell/Aspell.so
LD_RUN_PATH="/usr/lib" cc  -shared -Wl,-O1,--sort-common,--as-needed,-z,relro -L/usr/local/lib -fstack-protector Aspell.o  -o blib/arch/auto/Text/Aspell/Aspell.so 	\
   -laspell  	\
  
cc: internal compiler error: Segmentation fault (program collect2)
Please submit a full bug report,
with preprocessed source if appropriate.
See <https://github.com/archlinuxarm/PKGBUILDs/issues> for instructions.
Makefile:467: recipe for target 'blib/arch/auto/Text/Aspell/Aspell.so' failed
make: *** [blib/arch/auto/Text/Aspell/Aspell.so] Error 4
-> FAIL Installing Text::Aspell failed. See /root/.cpanm/work/1423512491.12855/build.log for details. Retry with --force to force install it.

WarheadsSE, 2015-02-09:
This is not a packaging issue, having to do with PKGBUILDs

closed 2015-02-09 by WarheadsSE

mauke, 2015-02-09:
Why was this bug closed?