archlinuxarm/PKGBUILDs #1099: package-name: chromium-pepper-flash

Poincare1, 2015-02-15:
Package: chromium-pepper-flash
Architecture: armv7
Device:  odroid-xu3

What it isn't doing:  Reflecting the current version of libpepflashplayer.so which is 16.0.0.305.

What it should be doing:  Be up-to-date.
 
Steps to reproduce:  installing current from this pkgbuild installs an old version and not the new version.  You can reproduce this by installing chromium=pepper-flash as seeing it is an old version of libpepflashplayer.so.

closed 2015-02-15 by kmihelich

Poincare1, 2015-02-15:
This follows the guidelines for submitting an issue.

Holzhaus, 2015-02-15:
No, it does not.

> The issue title must contain the package name and a short description of the problem:
>>package-name: short description

So the title of this should issue should be something like:
> alarm/chromium-pepper-flash: outdated version

instead of

> package-name: chromium-pepper-flash

Anyway, you'd better head over to the forums and discuss this issue here, because there might be another reason why kmihelich marked this as invalid.