archlinuxarm/PKGBUILDs #1103: nodejs: illegal instruction

breakpointsandy, 2015-02-18:
Package: nodejs-0.12.0-1
Architecture: armv6h
Device: raspberry pi

_What it isn't doing:_
When the node binary is executed, it gives the message __Illegal instruction (core dumped)__.
When run with the `--version` switch, it does show the correct version.

_What it should be doing:_
Executing as usual providing a prompt.

Steps to reproduce:
Run the `node` command with no arguments.

silverwind, 2015-03-01:
This was fixed in io.js 1.0.4, details [here](https://github.com/iojs/io.js/issues/283). I'd appreciate getting node.js replaced by io.js. They provide ARMv6 binaries too.

WarheadsSE, 2015-03-01:
We're aware that it has been addressed in io.js, and that it will be fixed in the upcoming version of node.js. We will not, however be replacing node.js. 

Also, io.js providing binaries does nothing for The Arch Way, and has no impact to the distribution.

WarheadsSE, 2015-03-04:
I have made comment over at the actual node.js Issues (https://github.com/joyent/node/pull/9068)

bendo, 2015-03-24:
0.12.1 was released ;)

yar0d, 2015-03-31:
A core dump occurs with this release too.

Package: community/nodejs 0.12.1-1
Architecture: ARMv7l PJ4
Device: Cubox - http://archlinuxarm.org/platforms/armv7/marvell/cubox

$ node --version
v0.12.1
$ node
Illegal instruction (core dumped)

silverwind, 2015-03-31:
@yar0d the fix hasn't landed yet in node, it targets 0.12.2 currently: https://github.com/joyent/node/pull/14193

WarheadsSE, 2015-04-01:
They have release 0.12.2, and still not fixed this. We will continue to push.

yar0d, 2015-04-08:
I hope that 0.12.3 will solve this :-)

silverwind, 2015-04-09:
We recently uncovered another armv6 SIGILL bug in v8: https://github.com/iojs/io.js/issues/1376

qjcg, 2015-09-02:
Package: nodejs-0.12.7-1
Architecture: armv7
Device: Odroid-XU3-lite, Odroid-C1

Running node to drop into the interpreter runs fine for me with the package version / hardware above.

```javascript
$ node
> console.log("Time to close issue #1103?");
Time to close issue #1103?
undefined
>
```
Is anyone still experiencing this issue?

silverwind, 2015-09-02:
@qjcg you're on ARMv7, this issue is about ARMv6.

node v4.0.0 will be available in a few days which will finally resolve this.

qjcg, 2015-09-02:
@silverwind, fair enough, I wasn't sure since @yar0d mentioned running into this on armv7 in his post above.

silverwind, 2015-09-02:
@qjcg Oh right, I missed that. For all I know, this never affected ARMv7 devices like RPi2 or Odroids. Maybe ARMv7l PJ4 is giving ARMv6-like `cpuinfo` and was hit by this bug as well, I'd suggest some testing with a recent io.js/node.js version on that device.

silverwind, 2015-09-08:
node v4.0.0 is now available which definitely fixes this.

kmihelich, 2015-09-17:
Fixed in 4.0.0-1.

closed 2015-09-17 by kmihelich