archlinuxarm/PKGBUILDs #1106: MySQL cluster build error

sebekk23, 2015-02-19:
Hi 

I'm trying to build MySQL cluster (with WITH_NDBCLUSTER) from sources using PKGBULD from here: https://aur.archlinux.org/packages/mysql but everything i can get is error like follows:
[code]
[ 64%] Built target ndbsched
Scanning dependencies of target ndbd
[ 64%] Building CXX object storage/ndb/src/kernel/CMakeFiles/ndbd.dir/main.cpp.o
[ 64%] Building CXX object storage/ndb/src/kernel/CMakeFiles/ndbd.dir/ndbd.cpp.o                                                                                                                                                                   
[ 64%] Building CXX object storage/ndb/src/kernel/CMakeFiles/ndbd.dir/angel.cpp.o                                                                                                                                                                  
In file included from /home/h3x0r/mysql-cluster/src/mysql-cluster-gpl-7.3.8/storage/ndb/include/debugger/EventLogger.hpp:24:0,
                 from /home/h3x0r/mysql-cluster/src/mysql-cluster-gpl-7.3.8/storage/ndb/src/kernel/angel.cpp:31:
/home/h3x0r/mysql-cluster/src/mysql-cluster-gpl-7.3.8/storage/ndb/include/kernel/signaldata/EventReport.hpp: In function 'void reportShutdown(const ndb_mgm_configuration*, NodeId, int, bool, bool, bool, Uint32, Uint32, Uint32)':
/home/h3x0r/mysql-cluster/src/mysql-cluster-gpl-7.3.8/storage/ndb/include/kernel/signaldata/EventReport.hpp:85:43: warning: 'theData.EventReport::eventType' is used uninitialized in this function [-Wuninitialized]
   eventType = (nodeId << 16) | (eventType & 0xFFFF);
                                           ^
/home/h3x0r/mysql-cluster/src/mysql-cluster-gpl-7.3.8/storage/ndb/src/kernel/angel.cpp:52:18: note: 'theData' was declared here
   Uint32 length, theData[25];
                  ^
[ 64%] Building CXX object storage/ndb/src/kernel/CMakeFiles/ndbd.dir/SimBlockList.cpp.o
Linking CXX executable ndbd
[ 64%] Built target ndbd
[ 64%] Building CXX object storage/ndb/src/kernel/vm/CMakeFiles/ndbsched_mt.dir/mt.cpp.o
/home/h3x0r/mysql-cluster/src/mysql-cluster-gpl-7.3.8/storage/ndb/src/kernel/vm/mt.cpp: In member function 'bool thr_send_threads::data_available()':
/home/h3x0r/mysql-cluster/src/mysql-cluster-gpl-7.3.8/storage/ndb/src/kernel/vm/mt.cpp:1249:9: error: 'rmb' was not declared in this scope
     rmb();
         ^
/home/h3x0r/mysql-cluster/src/mysql-cluster-gpl-7.3.8/storage/ndb/src/kernel/vm/mt.cpp: In member function 'void thr_send_threads::insert_node(NodeId)':
/home/h3x0r/mysql-cluster/src/mysql-cluster-gpl-7.3.8/storage/ndb/src/kernel/vm/mt.cpp:1412:9: error: 'wmb' was not declared in this scope
     wmb(); /* Ensure send threads see the change to m_first_node != 0 */
         ^
/home/h3x0r/mysql-cluster/src/mysql-cluster-gpl-7.3.8/storage/ndb/src/kernel/vm/mt.cpp: In member function 'NodeId thr_send_threads::get_node()':
/home/h3x0r/mysql-cluster/src/mysql-cluster-gpl-7.3.8/storage/ndb/src/kernel/vm/mt.cpp:1438:11: error: 'wmb' was not declared in this scope
       wmb(); /* Ensure send threads see the change to m_first_node == 0 */
           ^
/home/h3x0r/mysql-cluster/src/mysql-cluster-gpl-7.3.8/storage/ndb/src/kernel/vm/mt.cpp: In function 'void flush_write_state_other(thr_data*, thr_job_queue_head*, thr_jb_write_state*)':
/home/h3x0r/mysql-cluster/src/mysql-cluster-gpl-7.3.8/storage/ndb/src/kernel/vm/mt.cpp:2340:7: error: 'wmb' was not declared in this scope
   wmb();
       ^
/home/h3x0r/mysql-cluster/src/mysql-cluster-gpl-7.3.8/storage/ndb/src/kernel/vm/mt.cpp: In function 'void flush_write_state_other_wakeup(thr_data*, thr_job_queue_head*, thr_jb_write_state*)':
/home/h3x0r/mysql-cluster/src/mysql-cluster-gpl-7.3.8/storage/ndb/src/kernel/vm/mt.cpp:2397:7: error: 'wmb' was not declared in this scope
   wmb();
       ^
/home/h3x0r/mysql-cluster/src/mysql-cluster-gpl-7.3.8/storage/ndb/src/kernel/vm/mt.cpp: In function 'void flush_write_state_wakeup(const thr_data*, thr_data*, thr_job_queue_head*, thr_jb_write_state*)':
/home/h3x0r/mysql-cluster/src/mysql-cluster-gpl-7.3.8/storage/ndb/src/kernel/vm/mt.cpp:2422:9: error: 'wmb' was not declared in this scope
     wmb();
         ^
/home/h3x0r/mysql-cluster/src/mysql-cluster-gpl-7.3.8/storage/ndb/src/kernel/vm/mt.cpp: In function 'void flush_jbb_write_state(thr_data*)':
/home/h3x0r/mysql-cluster/src/mysql-cluster-gpl-7.3.8/storage/ndb/src/kernel/vm/mt.cpp:2452:7: error: 'wmb' was not declared in this scope
   wmb();
       ^
/home/h3x0r/mysql-cluster/src/mysql-cluster-gpl-7.3.8/storage/ndb/src/kernel/vm/mt.cpp: In function 'Uint32 link_thread_send_buffers(thr_repository::send_buffer*, Uint32)':
/home/h3x0r/mysql-cluster/src/mysql-cluster-gpl-7.3.8/storage/ndb/src/kernel/vm/mt.cpp:2813:13: error: 'rmb' was not declared in this scope
         rmb();
             ^
/home/h3x0r/mysql-cluster/src/mysql-cluster-gpl-7.3.8/storage/ndb/src/kernel/vm/mt.cpp: In function 'void pack_send_buffer(thr_data*, Uint32)':
/home/h3x0r/mysql-cluster/src/mysql-cluster-gpl-7.3.8/storage/ndb/src/kernel/vm/mt.cpp:3159:6: error: 'mb' was not declared in this scope
   mb();
      ^
/home/h3x0r/mysql-cluster/src/mysql-cluster-gpl-7.3.8/storage/ndb/src/kernel/vm/mt.cpp: In function 'void flush_send_buffer(thr_data*, Uint32)':
/home/h3x0r/mysql-cluster/src/mysql-cluster-gpl-7.3.8/storage/ndb/src/kernel/vm/mt.cpp:3225:7: error: 'wmb' was not declared in this scope
   wmb();
       ^
/home/h3x0r/mysql-cluster/src/mysql-cluster-gpl-7.3.8/storage/ndb/src/kernel/vm/mt.cpp: In member function 'virtual bool mt_send_handle::forceSend(NodeId)':
/home/h3x0r/mysql-cluster/src/mysql-cluster-gpl-7.3.8/storage/ndb/src/kernel/vm/mt.cpp:3268:8: error: 'mb' was not declared in this scope
     mb();
        ^
/home/h3x0r/mysql-cluster/src/mysql-cluster-gpl-7.3.8/storage/ndb/src/kernel/vm/mt.cpp: In function 'void try_send(thr_data*, Uint32)':
/home/h3x0r/mysql-cluster/src/mysql-cluster-gpl-7.3.8/storage/ndb/src/kernel/vm/mt.cpp:3302:8: error: 'mb' was not declared in this scope
     mb();
        ^
[... 21 lines cut ...]
make[2]: *** [storage/ndb/src/kernel/vm/CMakeFiles/ndbsched_mt.dir/mt.cpp.o] Error 1
CMakeFiles/Makefile2:3394: recipe for target 'storage/ndb/src/kernel/vm/CMakeFiles/ndbsched_mt.dir/all' failed
make[1]: *** [storage/ndb/src/kernel/vm/CMakeFiles/ndbsched_mt.dir/all] Error 2
Makefile:147: recipe for target 'all' failed
make: *** [all] Error 2
[/code] 

My makepkg.conf contains below flags:
- C,CPP, MAKE, DISTCC flags:
CFLAGS="-march=armv7ve -mfloat-abi=hard -mfpu=neon-vfpv4 -mcpu=cortex-a7 -mtune=cortex-a7 -O2 -pipe -fstack-protector --param=ssp-buffer-size=4"
CXXFLAGS="-march=armv7ve -mfloat-abi=hard -mfpu=neon-vfpv4 -mcpu=cortex-a7 -mtune=cortex-a7 -O2 -pipe -fstack-protector --param=ssp-buffer-size=4"
CPPFLAGS="-D_FORTIFY_SOURCE=2 -march=armv7ve -mfloat-abi=hard -mfpu=neon-vfpv4 -mcpu=cortex-a7 -mtune=cortex-a7 -O2 -pipe -fstack-protector --param=ssp-buffer-size=4"
MAKEFLAGS="-j11"
DISTCC_HOSTS="r1.dom markone.dom marktwo.dom vulcanone.dom vulcantwo.dom"

Env is like follows:
- OS Linux (ArmArchLinux)
- Kernel: Linux r1 3.4.104+ #1 SMP PREEMPT Thu Jan 8 15:40:40 CET 2015 armv7l GNU/Linux
- gcc
gcc (GCC) 4.9.2 20141224 (prerelease)
Copyright (C) 2014 Free Software Foundation, Inc.
This is free software; see the source for copying conditions. There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

It has something to do with memory barrier but i can not figure out what i should include and where to make it work with ARM Kernel Sources.

closed 2015-02-19 by kmihelich

sebekk23, 2015-02-19:
Why you marked this thread as invalid ?

WarheadsSE, 2015-02-19:
Because mysql was replaced by mariadb quite a while ago. It is not longer an official package from Arch Linux or Arch Linux ARM.

Thus, per guidelines that this Issue tracker is only for PKGBUILDs within the tree, this is entirely invalid.