archlinuxarm/PKGBUILDs #1108: xf86-video-fbdev: driver not loading

pjaneck, 2015-02-20:
Package: xf86-video-fbdev 0.4.4-4
Architecture: armv7h
Device: Samsung Chromebook XE303C12

What it isn't doing: loading the driver
dmesg gives:
[drm:exynos_drm_gem_dumb_create] ERROR contig buffer allocation not supported
I don't know if this message is related.
When trying to start the xserver manualy with startx the log says:
Fatal server error: AddScreen/ScreenInit failed for driver 0

What it should be doing: loading the driver

Steps to reproduce: just install xf86-video-fbdev 0.4.4-4 and xorg-server 1.17.1-1 and reboot

pepedog, 2015-02-20:
Is xorg-xinit installed? A typical use would be xinit /usr/bin/lxde
The forum is the right place for this as nothing is wrong with package

closed 2015-02-20 by pepedog

pjaneck, 2015-02-23:
I allready put this issue in the forum and it is confirmed by others. Is the github just for errors in the package build or also for bugs in the package code itself?
Btw. xorg-xinit is installed. 
I will try to run the xserver via xinit later this day, but I think this will lead to the same problem.