archlinuxarm/PKGBUILDs #1109: rrdtools: floating point exception

ineiti, 2015-02-20:
Package: rrdtools
Architecture: armv5
Device: Dreamplug (NON-dt kernel)

What it isn't doing:
it fails upon creating an output image

What it should be doing:
create the output image

Steps to reproduce:
When I run rrdtools twice to create an image, I get the following error:

```
gettimeofday({1424428357, 714398}, NULL) = 0
access("/usr/share/fonts/TTF/DejaVuSansMono.ttf", R_OK) = 0
brk(0xcd000)                            = 0xcd000
mmap2(NULL, 663552, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_ANONYMOUS, -1, 0) = 0xb5f84000
tgkill(2786, 2786, SIGFPE)              = 0
--- SIGFPE {si_signo=SIGFPE, si_code=SI_TKILL, si_pid=2786, si_uid=0} ---
+++ killed by SIGFPE (core dumped) +++
Floating point exception (core dumped)
```

If I follow the instructions in http://archlinuxarm.org/forum/viewtopic.php?f=15&t=8003 - that is, if I delete the directory ```/var/cache/fontconfig```, I can use rrdtools. The author in there also tried with a clean installation and got the same results.

Most probably I should file an error against another package, but I don't know if it's ```fontconfig``` or something else that is creating the error.

ineiti, 2015-02-20:
I just tried a simple

```
convert -pointsize 15 -font /usr/share/fonts/TTF/DejaVuSansMono.ttf label:"The quick brown fox" font.png
```

which doesn't give an error.

kmihelich, 2015-02-20:
List the steps to reproduce this.

ineiti, 2015-02-20:
Le 20/02/2015 16:11, Kevin Mihelich a écrit :
> List the steps to reproduce this.

The following works on a Smileplug - armv7:

```
rm traffic.rrd

rrdtool create traffic.rrd --step 10  DS:hostdp:COUNTER:60:0:1000000 \
DS:hostmbp:COUNTER:60:0:1000000 RRA:AVERAGE:0.5:1:600 \
RRA:MIN:0.5:1:600 RRA:MAX:0.5:1:600 RRA:AVERAGE:0.5:6:1440 \
RRA:MIN:0.5:6:1440 RRA:MAX:0.5:6:1440 RRA:AVERAGE:0.5:60:1728 \
RRA:MIN:0.5:60:1728 RRA:MAX:0.5:60:1728

for a in 1 2; do
rrdtool graph -u 1000000 /tmp/image.png --start -1800 --step 10 -a PNG \
-t 'Traffic to internet' --vertical-label \
'bps' -w 600 -h 150 -r   DEF:hostdp=traffic.rrd:hostdp:AVERAGE \
DEF:hostmbp=traffic.rrd:hostmbp:AVERAGE   CDEF:bhostdp=hostdp,8,* \
CDEF:bhostmbp=hostmbp,8,* LINE1:0  AREA:bhostdp#FF9999:DreamPlug:STACK \
AREA:bhostmbp#99FF99:MacBook_Pro:STACK
done
```

But it fails on a Dreamplug - armv5. This very example doesn't even work 
with deleting /var/cache/fontconfig, so this might be another problem 
altogether...

kmihelich, 2015-02-21:
cairo 1.14.0-2.1 should fix this, pacman -Syu.

closed 2015-02-21 by kmihelich