archlinuxarm/PKGBUILDs #1111: package-name: short description

hp4, 2015-02-21:
Package: ntfs-utils
Architecture: armv6h
Device: Raspberry Pi

What it isn't doing: mount.nfs doesn't work after last update to 1.3.2-2

What it should be doing: mount nfs devices as in version 1.3.1.1

Steps to reproduce: issue comand  "mount.nfs -v  ..." yields 
"Program/version mismatch" (version seems to be corrupted)

====================================================================
mount.nfs: timeout set for Sat Feb 21 16:16:34 2015
mount.nfs: trying text-based options 'nolock,vers=4.2,addr=192.168.178.3,clientaddr=192.168.178.9'
mount.nfs: mount(2): Protocol not supported
mount.nfs: trying text-based options 'nolock,vers=4.1,addr=192.168.178.3,clientaddr=192.168.178.9'
mount.nfs: mount(2): Protocol not supported
mount.nfs: trying text-based options 'nolock,vers=4.0,addr=192.168.178.3,clientaddr=192.168.178.9'
mount.nfs: mount(2): Protocol not supported
mount.nfs: trying text-based options 'nolock,addr=192.168.178.3'
mount.nfs: prog 100003, trying vers=3065173208, prot=6
mount.nfs: trying 192.168.178.3 prog 100003 vers 3065173208 prot TCP port 2049
mount.nfs: portmap query retrying: RPC: Program/version mismatch
mount.nfs: prog 100003, trying vers=3065173208, prot=17
mount.nfs: trying 192.168.178.3 prog 100003 vers 3065173208 prot UDP port 2049
mount.nfs: portmap query failed: RPC: Program/version mismatch
mount.nfs: Protocol not supported

====================================================================

while same command immediately after rollback to previous nfs-utils-1.3.1-1 yields

mount.nfs: timeout set for Sat Feb 21 16:17:44 2015
mount.nfs: trying text-based options 'nolock,vers=4,addr=192.168.178.3,clientaddr=192.168.178.9'
mount.nfs: mount(2): Protocol not supported
mount.nfs: trying text-based options 'nolock,addr=192.168.178.3'
mount.nfs: prog 100003, trying vers=3, prot=6
mount.nfs: trying 192.168.178.3 prog 100003 vers 3 prot TCP port 2049
mount.nfs: prog 100005, trying vers=3, prot=17
mount.nfs: trying 192.168.178.3 prog 100005 vers 3 prot UDP port 33241

WarheadsSE, 2015-02-21:
Invalid.

WarheadsSE, 2015-02-21:
Less invalid

kmihelich, 2015-02-21:
Usage issue, not a package bug.

closed 2015-02-21 by kmihelich