archlinuxarm/PKGBUILDs #1114: Kodi-rbp: Alsa disabled

leonardder, 2015-02-24:
Package: Kodi-RBP
Architecture: ARMV7H
Device: Raspberry Pi 2

What it isn't doing:
Kodi-rbp has alsa explicitly disabled. Therefore, external USB soundcards aren't supported by Kodi.

What it should be doing:
Alsa should be enabled --enable-alsa to support alsa output for external soundcards.

Steps to reproduce:
Go to settings, system and try to select an audio output. However an external soundcard is attached to the device, Kodi doesn't detect it because alsa support isn't built in.

kmihelich, 2015-02-24:
Why is this not a pull request?

leonardder, 2015-02-24:
Because I'm a blind user and Githubs pull request system is somehwat accessible. I however added two commits to http://github.com/leonardder/PKGBUILDs master branch, one for kodi-rbp and another for kodi-rbp-git to keep them in sync.

jolan, 2015-02-26:
I just did PR#1116 before seeing this.  I only did stable since that's all I've tested so far.

commit d873dc77 (2015-02-28): alarm/kodi-rbp-git to 15.20150228-1

fixes #1114
closed 2015-02-28 by kmihelich