archlinuxarm/PKGBUILDs #1115: Kodi-rbp: Keyboard letters not recognized

leonardder, 2015-02-24:
Package: Kodi-RBP
Architecture: ARMV7H
Device: Raspberry Pi 2

What it isn't doing:
I have an external keyboard attached to my Banana Pi to operate Kodi. It is currently not possible to use the letter keys on the keyboard to navigate in lists or input in fileds (such as device name). However, pressing the p key for play works just fine.

What it should be doing:
Inputting letters in input fields and with quick navigation in lists should work correctly.

Steps to reproduce:
1. Go to settings>services>general>device name. It is currently not possible to input a new name there
2. Open a folder in the music library and try to navigate using shift+letter keys, i.e. shift+c to move to the classical music folder. Quick navigation doesn't work.

I've tested this on OpenElec with exactly the same settings, and it works as expected there.

kmihelich, 2015-02-24:
Unsupported device.

closed 2015-02-24 by kmihelich

leonardder, 2015-02-24:
Hmm, ok. Could you please give a more detailed reason? What device is unsupported? Why should something that works on another OS be unsuported in this package?

kmihelich, 2015-02-24:
> I have an external keyboard attached to my Banana Pi to operate Kodi.

leonardder, 2015-02-24:
Ah, should have double checked my message. Fixed the initial post accordingly, the device in question is a Raspberry Pi 2.

dhead666, 2015-02-24:
Try chmod /dev/tty0 to 0777

kmihelich, 2015-02-24:
In that case, usage issues belong in the forum.  Still invalid.