archlinuxarm/PKGBUILDs #1128: package-name: xorg-server

x11tete11x, 2015-03-07:
Package: xorg-server (i think)
Architecture: armv6h
Device: Raspberry Pi B+

What it isn't doing: Xorg segfault when i move the mouse (Microsoft Wireless Optical Desktop 1000)

What it should be doing: Should works when i move the mouse

Steps to reproduce:
1) startx (in my case LXDE starts)
2) move the mouse (you will se an Xorg crash by a segfault)

logs and more info: http://archlinuxarm.org/forum/viewtopic.php?f=59&t=8583

kmihelich, 2015-03-08:
"I think" is not a valid qualification.  When you have identified a specific bug in a specific package, let us know.

closed 2015-03-08 by kmihelich

x11tete11x, 2015-03-08:
:S but how i know what it's exactly wrong? there is a bug in somewhere.. i cant tell you if it's exactly on this package, or in udev or what..