archlinuxarm/PKGBUILDs #1131: Errors when using FIQ

Mentalleap, 2015-03-09:
Package: Kernel?

Architecture: Arch Linux Arm, latest Version (effects also eaylier versions), using latest Bluez and pulseaudio

Device: Raspberry B

What it isn't doing:
Using Bluetooth Dongles the system reports a lot of "Transfer to device X endpoint 0xY failed - FIQ reported NYET" errors. 
Using dwc_otg.fiq_enable=0 in cmdline.txt will fix the problem.

Tried 2 dongles. One very old BT v2.1 and a new v4.0 dongle running either on hub or directly on raspi. running directly on raspi may cause less errors but there are still errors.

This causes some other error. I'm running mpd as a UPnP renderer to play to alsa and then logitechmediaserver to distribute this to other clients. The "FIQ Error" causes a "broken pipe" error causing the connection to the client beeing closed. Even if the bluetooth device is just connected to USB and there is no BT connection to i.e a mobile.

Changing CPU speed does no change anything. Tried all settings from 700 to 1000MHz

What it should be doing:
Should not cause transfer errors or at least not broken pipe errors

Steps to reproduce:
The transfer error is easy to reproduce. Just plug in a usb bt device.

kmihelich, 2015-03-09:
We don't develop the kernel, we only package it.  You need to report this to the correct issue tracker.

closed 2015-03-09 by kmihelich

Mentalleap, 2015-03-09:
Sorry, did not know, that the settings in cmdline.txt are from someone else.