archlinuxarm/PKGBUILDs #1138: Kodi running from systemd does not allow restart or shutdown

graysky2, 2015-03-19:
With a recent update, I am unable to shutdown or restart kodi-rbp from the GUI.  In the past, these options were available from the "power" button within the GUI itself.  For some reason, now, only "exit" is available from this button.

Following WarheadsSE's suggestion in [this thread](http://archlinuxarm.org/forum/viewtopic.php?f=60&t=8631), I can verify that indeed, the kodi user is not listed in the output of loginctl indicating that it is not given a proper seat which is likely to blame.  Note that at least one other user in the thread I linked is also experiencing this on a RPi2 so I believe it to be systemic.

The kodi user created by the package for info:
```
% id kodi
uid=997(kodi) gid=997(kodi) groups=997(kodi),6(disk),90(network),91(video),92(audio),93(optical),95(storage),98(power)
```
Package version: alarm/kodi-rbp 14.1-4

closed 2015-03-19 by kmihelich

graysky2, 2015-03-19:
kmihelich - why is my report invalid?

moonman, 2015-03-19:
See contributing guidlines.

graysky2, 2015-03-19:
Sorry, I was unaware of the nice template shown here: https://github.com/archlinuxarm/PKGBUILDs/blob/master/CONTRIBUTING.md

I will open a new issue formatted correctly.