archlinuxarm/PKGBUILDs #1140: kodi-rbp: included systemd service does not allow kodi user to restart or shutdown

graysky2, 2015-03-19:
Package: kodi-rbp 14.1-4
Architecture: both armv6 and armv7
Device: RPi and RPi2

What it isn't doing: Interacting with systemd to allow Kodi to control system restart or shutdown.

What it should be doing: Allowing the system to restart or to shutdown from the Kodi GUI.

Steps to reproduce:
1) Install this package on RPi or RPi2
2) Start the kodi service
3) Try to restart or shutdown from the Kodi GUI (arrow down to the power button icon and hit ENTER)

graysky2, 2015-03-19:
I am proposing the fix in PR#1141 which I have tested on a my RPi.

kmihelich, 2015-03-19:
I am waiting for a very good reason why you're opening issues you have a fix for.

graysky2, 2015-03-19:
@kmihelich - When I opened the issue, I had no idea for a fix.  Since opening it, I believe my PR is an option.  Perhaps someone with more knowledge on the dev team will see my proposal and think of something better.

kmihelich, 2015-03-19:
No idea?  Really?
https://github.com/graysky2/kodi-standalone-service/blob/master/init/kodi.service

closed 2015-03-19 by kmihelich

graysky2, 2015-03-19:
Yes, I maintain that package for Arch.  I didn't make the connection until Warhead posted in this thread suggesting the seat issue, this is dawned on me you guys might be doing this differently and I more or less merged the Arch package into your package.

http://archlinuxarm.org/forum/viewtopic.php?f=60&t=8631#p46011

graysky2, 2015-03-19:
Also, someone with a fresh install needs to test my PR to verify that I have correctly captured all the needed deps.  As I said, it works on my RPi but it has more packages installed than the base, so additional ones might be needed.

graysky2, 2015-03-20:
https://github.com/archlinuxarm/PKGBUILDs/pull/1142

starquake, 2015-06-05:
I did an install of kodi-rbp yesterday. I installed kodi first. That didn't work. Then I found out there's this kodi-rbp package. I installed that and it works. (It even looks like the performance in the menu is better than openelec.) Anyway. I can't shutdown or reboot. I also installed udisks but drives are not mounted automatically (might be a separate issue) 

You needed someone with a fresh install. Can you help me troubleshoot this problem? :-)

Do you think I'm missing a dependency or do you think I messed up some config file by installing the wrong kodi version first?

graysky2, 2015-06-05:
Install polkit.

starquake, 2015-06-05:
Is installing polkit with `pacman -S polkit` enough or do I need to do some more configuring?

graysky2, 2015-06-05:
Just install polkit and restart kodi.

starquake, 2015-06-06:
As I suspected polkit is already installed. So that won't fix it. I'm gonna try to install from scratch because I might have done some funky stuff trying to get kodi working (instead of kodi-rpb) and report back. Thanks for helping.

starquake, 2015-06-08:
Reinstall from scratch fixed it! Too bad I have no idea what was causing it.