archlinuxarm/PKGBUILDs #1143: wpa_supplicant: does not initialize RTL8188CUS 802.11n WLAN Adapter

gituser6, 2015-03-21:
Package: wpa_supplicant 2.3-1
Architecture: armv6h
Device: Raspberry Pi

What it isn't doing: Does not interact with this hardware while other tools such as netctl's wifi-menu can.

What it should be doing: It should bring up the device and allow connections to networks.

Steps to reproduce:

1) Plug in USB wi-fi hardware
2) make `/etc/wpa_supplicant/example.conf` that contains this
```
ctrl_interface=/run/wpa_supplicant
update_config=1
```
3) Run this command as root user
```
wpa_supplicant -i wlan0 -c /etc/wpa_supplicant/example.conf
```

The error occurs here and is
```
Successfully initialized wpa_supplicant
nl80211: Driver does not support authentication/association or connect commands
wlan0: Failed to initialize driver interface
```

If I use `wifi-menu -o` it does throw some errors but the hardware comes up and allows connections.

Errors by wifi-menu
```
Scanning for networks... ioctl[SIOCSIWAP]: Operation not permitted
ioctl[SIOCSIWENCODEEXT]: Invalid argument
ioctl[SIOCSIWENCODEEXT]: Invalid argument
done
```

WarheadsSE, 2015-03-21:
Not a problem with that package, as the error states.

```nl80211: Driver does not support authentication/association or connect commands```

closed 2015-03-21 by WarheadsSE

gituser6, 2015-03-21:
I thought that too but if I plug the very same hardware into my laptop and repeat the identical steps, it works without these errors and I am able to connect. I think it is something specific to our package for armv6h since x86_64 works.

gituser6, 2015-03-21:
I booted into raspbian-wheezy with the same hardware and was able to successfully use wpa_supplicant there so I believe the problem is with our package.

On raspbian:
```
$ sudo  wpa_supplicant -B -i wlan0 -c /etc/wpa_supplicant/example.conf
rfkill: Cannot open RFKILL control device
ioctl[SIOCSIWAP]: Operation not permitted
ioctl[SIOCSIWENCODEEXT]: Invalid argument
ioctl[SIOCSIWENCODEEXT]: Invalid argument
```

Those are different warning messages that do not keep wpa_supplicant from running
```
$ ps aux |grep wpa
root      2433  0.0  0.5   5656  2456 ?        Ss   19:45   0:00 wpa_supplicant -B -i wlan0 -c /etc/wpa_supplicant/example.conf
```

Further, I can now connect to it with wpa_cli and connect to my router
```
$ sudo wpa_cli
wpa_cli v1.0
Copyright (c) 2004-2012, Jouni Malinen <j@w1.fi> and contributors

This program is free software. You can distribute it and/or modify it
under the terms of the GNU General Public License version 2.

Alternatively, this software may be distributed under the terms of the
BSD license. See README and COPYING for more details.

Selected interface 'wlan0'

Interactive mode

> scan
OK
<3>CTRL-EVENT-SCAN-RESULTS 
<3>WPS-AP-AVAILABLE 
> scan_results
bssid / frequency / signal level / flags / ssid
74:d0:2b:6c:ec:18	2437	87	[WPA2-PSK-CCMP][ESS]	two chicks one router
11:5b:d1:9e:8c:60	2412	46	[WPA-PSK-TKIP+CCMP][WPA2-PSK-TKIP+CCMP][ESS]	ATT7
40:b7:f3:14:c3:50	2462	10	[WPA-PSK-TKIP+CCMP][WPA2-PSK-TKIP+CCMP][ESS]	ATT4M2
40:c3:97:0b:b7:28	2427	26	[WPA-PSK-TKIP+CCMP][WPA2-PSK-TKIP+CCMP][ESS]	GO_SOCKS
30:62:23:83:93:e1	2412	46	[WPA-PSK-TKIP+CCMP][WPA2-PSK-TKIP+CCMP][ESS]	Sharkbite
78:96:84:30:21:c0	2457	10	[WPA-PSK-TKIP+CCMP][WPA2-PSK-TKIP+CCMP][ESS]	notyournetwork
```

graysky2, 2015-03-22:
I am following your thread on rpi forums.  What version of wpa_supplicant does raspbian use?  Arch ARM is using 2.3 currently.  That could be to blame for the differences if the versions are different.  Please find out and consider compiling the same version on Arch ARM for a true apples-to-apples comparison.

gituser6, 2015-03-22:
I don't know how to build such an old version. Can you help?

graysky2, 2015-03-22:
Browse to arch packages and find wpa_supplicant. Then click "view changes" and pull the last known good version [1.0-xxx build](https://projects.archlinux.org/svntogit/packages.git/commit/trunk?h=packages/wpa_supplicant&id=41a261cc0b926fdaab953e489a8700d759835d8d).

It looks like there is a patch which I found by [googling](https://github.com/StarchLinux/ports/blob/master/core/wpa_supplicant/hostap_allow-linking-with-libnl-3.2.patch)