archlinuxarm/PKGBUILDs #1155: webkit2gtk: failing to render Web pages and then seems to grab all RAM

kaihendry, 2015-04-15:
Package: webkit2gtk
Architecture: webkit2gtk
Device: Rpi2

What it isn't doing: Rendering Web pages with a style sheet

What it should be doing: Rendering Web pages with a style sheet

Steps to reproduce: `epiphany http://google.com`
You could also use https://aur.archlinux.org/packages/surf2/ which drives webkit2gtk without the cruft.

On my x86_64 laptop, webkit2gtk 2.8.0-2 is fine. Another thing I noticed that it is able to render a Web page WITHOUT a style sheet. e.g. http://archpi.dabase.com/

kmihelich, 2015-04-15:
If this isn't a webkit2gtk bug, why is this here?  And when did webkit2gtk become an ARM architecture?

kaihendry, 2015-04-15:
Whoa, you're a pedantic fellow! I correct the paste mishtake.

I can't isolate the bug. It has to go somewhere and it affects anything depending on /usr/lib/libwebkitgtk-1.0.so. I wonder why it's `/usr/lib/libwebkit2gtk-4.0.so` on Architecture: x86-64. Ho hmm

kmihelich, 2015-04-15:
When you find a bug in a specific package, let the upstream project know, then let us know.

closed 2015-04-15 by kmihelich

kaihendry, 2015-04-15:
Why did you close? There is still an outstanding issue in **alarm**. Shouldn't this bug be closed once it's addressed?

WarheadsSE, 2015-04-15:
Because this is not a packaging issue.

kaihendry, 2015-05-18:
How do I get the CMakeCache.txt for the Webkit2 developers? https://bugs.webkit.org/show_bug.cgi?id=143817#c1

kaihendry, 2015-10-29:
From 2.10.1 Emanuele says Webkit2 in Rpi2 needs to be built with **-DENABLE_GL=OFF**. Could https://github.com/archlinuxarm/PKGBUILDs/blob/master/extra/webkit2gtk/PKGBUILD please be updated? 
I wish I could test this myself, but I don't have a sane build system.

Details are upon https://bugs.webkit.org/show_bug.cgi?id=143817#c4

I've tried to summarise the back story upon http://dabase.com/blog/Webkit_on_Rpi2/