archlinuxarm/PKGBUILDs #1156: kodi-rbp: segmentation fault error

kovscer, 2015-04-15:
**Package**: kodi-rbp-14.2.1
**Architecture**: armv6h
**Device**: raspberrypi

**What it isn't doing**:
Kodi doesn't start.

**What it should be doing**:
Start normally without producing a segmentation fault error.

**Steps to reproduce**:
Fresh install of Archlinuxarm with up to date packages. Just start kodi service and you will see that nothing happened except crash log and core dump generation.

If you need more information, don't hesitate to ask me.

kmihelich, 2015-04-15:
Can't reproduce.

closed 2015-04-15 by kmihelich

moonman, 2015-04-15:
I have had the same problem with arm-mem-git installed. Remove that package
and try kodi again.

kovscer, 2015-04-16:
I had the same problem thank you moonman for your comment. But what i can't understood its why its not working anymore with arm-mem-git installed.