archlinuxarm/PKGBUILDs #1157: linux-wandboard: unable to build due to ssl problem

mihadyuk, 2015-04-16:
Package: linux-wandboard
Architecture: archv7
Device: wandboard

What it isn't doing: when I try to execute makepkg -s, I get the following error:
Here is output.
[user@wandboard linux-wandboard]$ makepkg -s
==> Making package: linux-wandboard 3.10.17-5 (Wed Dec 31 17:59:57 MST 1969)
==> Checking runtime dependencies...
==> Checking buildtime dependencies...
==> Retrieving sources...
  -> Downloading be8d6872b5eb4c94c15dac36b028ce7f60472409.tar.gz...
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0curl: (60) SSL certificate problem: certificate is not yet valid
More details here: http://curl.haxx.se/docs/sslcerts.html

curl performs SSL certificate verification by default, using a "bundle"
 of Certificate Authority (CA) public keys (CA certs). If the default
 bundle file isn't adequate, you can specify an alternate file
 using the --cacert option.
If this HTTPS server uses a certificate signed by a CA represented in
 the bundle, the certificate verification probably failed due to a
 problem with the certificate (it might be expired, or the name might
 not match the domain name in the URL).
If you'd like to turn off curl's verification of the certificate, use
 the -k (or --insecure) option.
==> ERROR: Failure while downloading be8d6872b5eb4c94c15dac36b028ce7f60472409.tar.gz
    Aborting...

What it should be doing:
It should get sources successfully and then start making pakage.

Steps to reproduce:
1. Power on the wandboard.
2. Open console and connect via ssh to the board as user. I used ssh user@192.168.1.10.
3. navigate to folder with pkgbuild by issuing cd path/to/pkgbuild.
4. exec makepkg -s command.

moonman, 2015-04-16:
`linux-wandboard 3.10.17-5 (Wed Dec 31 17:59:57 MST 1969)
 SSL certificate problem: certificate is not yet valid`

Not a packaging problem.

closed 2015-04-16 by moonman

mihadyuk, 2015-04-16:
Could you help me. How can I disable ssl verifying in the pkgbuild?

moonman, 2015-04-16:
I highlighted the important parts of your log so by now it should be pretty obvious that you need to set correct date and time. Please read the rules for posting issues. Usage issues belong on the forum, not here!

mihadyuk, 2015-04-16:
Thanks for your help. I've guessed already. Now time is configured and the problem is over.