archlinuxarm/PKGBUILDs #1161: nginx 1.6.3-1: proxy_pass not working

JB26, 2015-04-16:
Package: nginx 1.6.3-1
Architecture: armv6
Device: Raspberry Pi

What it isn't doing:
The worker process dies when trying to connect to a service through nginx reverse_proxy. No data is served.
This function is working normally on x86_64.

What it should be doing:
The worker process shouldn't die and nginx should send data.

Steps to reproduce:
Please see: [Forum Post](http://archlinuxarm.org/forum/viewtopic.php?f=57&t=8730)

Workaround:
Downgrade to nginx 1.6.2-2

kmihelich, 2015-04-16:
Have you filed a bug with nginx? We are not the nginx developers.

JB26, 2015-04-16:
I now have: http://trac.nginx.org/nginx/ticket/748#ticket
I weren't sure where to post this bug because it seems to be only happening on arm.
Sorry if this was the wrong place to post this.
Thanks

ajs124, 2015-04-17:
I have the same problem, but it doesn't seem to be related to the nginx version.
I use a patched version of nginx (only one addition module, shouldn't be relevant) on my raspberry pi and after 1.6.3 was released on the repos, I upgraded my version and it broke (workers  segfaulting + server not sending data).
So naturally, I tried downgrading, but since I didn't have the old package around, I compiled the previous version (1.6.2) and it behaves the same. Then I tried 1.7.12, which also didn't work. So I compiled a debug build (```--with-debug``` and ```options=('debug' '!strip')``` in the PKGBUILD) to debug the problem. The debug build doesn't have this problem. So yeah… no idea what to do now, maybe the nginx devs have an idea.
What changed since the 1.6.2 build in the Arch Linux ARM stack? New compiler? New kernel? New libc?

JB26, 2015-04-17:
When I first got this bug I compiled nginx myself and got the same bug. Today I compiled nginx again. The new build is 60kb bigger and the bug is gone…
According to my logs the only packages that changed were some php and python packages and I rebooted.
I tried downgrading the kernel but I can't reproduce the faulty build.
So `pacman -Syu; reboot` and rebuilding nginx seems to fix this bug?

daniel-j, 2015-04-18:
Recompiled on the pi solved it for me

bartbes, 2015-04-18:
A rebuild (with debug symbols) solved the same problem for me too.

I can also add that it segfaults on uwsgi_pass, which is how I initially found this.

As for the build date, I recompiled (successfully) on Wed 15 Apr 2015 11:55:50 CEST according to pacman, so that was before @JB26 first recompiled.

kmihelich, 2015-04-18:
Test this package:
`pacman -U http://archlinuxarm.org/builder/testing/nginx-1.6.3-1-armv6h.pkg.tar.xz`

ajs124, 2015-04-18:
Seems to work for me. What's the difference between this and what's currently in the repos?

kmihelich, 2015-04-18:
Building with -O1 instead of -O2.

bartbes, 2015-04-18:
Unfortunately doesn't seem to work for me with uwsgi_pass (proxy_pass untested).

JB26, 2015-04-18:
Doesn't work for me either

JB26, 2015-04-18:
I think I found it.
Add ´makedepends=('hardening-wrapper')´ to the PKGBUILD.
I oversaw this because I only checked the differences in the PKGBUILDs on archlinux.org and didn't check for differences on github.com/archlinuxarm/PKGBUILDs and archlinux.org.

bartbes, 2015-04-18:
I used the arch pkgbuild (containing the hardening-wrapper dependency) when I was compiling it, so that definitely could be the case.

kmihelich, 2015-04-18:
Try this one (same file, new package):
`pacman -U http://archlinuxarm.org/builder/testing/nginx-1.6.3-1-armv6h.pkg.tar.xz`

bartbes, 2015-04-18:
Still segfaults, sadly.

JB26, 2015-04-18:
This one works for me :)
@bartbes Make sure pacman didn't install a cached version of the package. I needed to manually grab the new build with wget because pacman was trying to install the older version.

bartbes, 2015-04-18:
You were right, it appears it did grab a cached copy, it does work, even with uwsgi_pass.

closed 2015-04-18 by kmihelich

bartbes, 2015-04-18:
Thanks for the swift fix!

JB26, 2015-04-18:
Thanks for the fix.
I added a comment to the nginx bug report.

commit e0972be7 (2015-04-18): extra/nginx to 1.6.3-1.1

fixes #1161