archlinuxarm/PKGBUILDs #1164: weston-rpi 1.7.0-1: fails to load libinput.so.7, it should be looking for libinput.so.10

pisto, 2015-04-20:
Package: weston-rpi 1.7.0-1
Architecture: armv7l
Device: raspberry pi 2 model b
What it isn't doing: work, at all.
What it should be doing: work, at all.
Steps to reproduce: install weston-rpi, log in into a virtual terminal, launch weston-launch. It fails to find libinput.so.7(LIBINPUT_0.8.0), but no package provides it. It seems there's only libinput.so.10(LIBINPUT_0.12.0) from package libinput 0.13.0-2.

kmihelich, 2015-04-20:
@plfiorini this is yours to fix.

plfiorini, 2015-04-20:
@kmihelich libinput has changed apparently in an incompatible way, rebuild weston-rpi with the new libinput

kmihelich, 2015-04-20:
Submit a PR with the necessary updates.

pisto, 2015-05-07:
ping

kmihelich, 2015-05-07:
@plfiorini if you have abandoned this package, please inform us so it can be deleted.

plfiorini, 2015-05-07:
@kmihelich I haven't abandoned this package. Booting the RPi2 so I can rebuild weston.

plfiorini, 2015-05-07:
https://github.com/archlinuxarm/PKGBUILDs/pull/1183

closed 2015-05-07 by kmihelich