archlinuxarm/PKGBUILDs #1168: package-name: systemd (219-5 + 219-5.1)

janmulder, 2015-04-23:
Package: systemd
Architecture: armv7h
Device: Raspberry Pi 2 

What it isn't doing: Bring up the IPv4 network on boot. Being attached to a IPv6 capaable router, the IPv6 interface is started correctly. 

What it should be doing: Bring up the IPv4 network on boot (and the IPv6 if not disabled)

Steps to reproduce: Start with a22fc505df09e9ed0de5fee64afba5d8  ArchLinuxARM-2015.04-rpi-2-rootfs.tar.gz (the current latest). Boot it, and check the network connectivity using the DHCP assigned addresses (both IPv4 and 6). Notice that it contains systemd 218. Now, pacman -Syu, resulting in (among others) systemd 219-5.1 Reboot, and check the network connectivity again. IPv4 does not work, IPv6 does. 

In addition: I downgraded the following packages (to 218-2):
libsystemd
systemd
systemd-sysvcompat

and the IPv4 and IPv6 are started correctly on boot.

kmihelich, 2015-04-23:
Can't reproduce.

closed 2015-04-23 by kmihelich

WarheadsSE, 2015-04-23:
I can't reproduce this either. The DHCP client at the heart of systemd-networkd (it does not use dhcpc) is becoming very RFC compliant. It is possible that your setup is having issues with that due to timeouts, non-compliance, or other means.