archlinuxarm/PKGBUILDs #1176: uboot-cu* installs on /dev/mmcblk0

BlackLotus, 2015-04-27:
You can install archlinux on the sdcard or an usb disk or a sata disk or and that's the default setup for many.
The package does:
    dd if=/boot/u-boot-sunxi-with-spl.bin of=/dev/mmcblk0 bs=1024 seek=8
which is not universally useable and should be changed dynamically (perhaps using /proc/cmdline to find out which partition to use)

WarheadsSE, 2015-04-27:
CONTRIBUTING.md

Also: I am afraid you're wrong here. The boot0 code of the Allwinner SoC will only look to the NAND or eMMC/SD @ mmcblk0. Even if you install to another location, the boot loader must be atthis location.

closed 2015-04-27 by WarheadsSE

BlackLotus, 2015-04-27:
Sorry next time I open a bug report I will use the contributing.md pattern.
And yup oops thought this was the sdcard.Seems like it is really the default location.
Sorry about that.

BlackLotus, 2015-04-27:
Sorry next time I open a bug report I will use the contributing.md pattern.
But I still believe that uboot gets only installed on an sdcard, but not on the nand. I will take a closer look at it soon.