archlinuxarm/PKGBUILDs #1190: kodi-rbp: no keyboard access for filling text fields in kodi

joaodriessen, 2015-05-17:
Package: kodi-rbp
Architecture: armv7h
Device: Raspberry Pi 2

What it isn't doing: not able to type into kodi, keyboard works fine for navigation though.

What it should be doing: keyboard should be available for typing, and not just for navigation/shortcuts

Steps to reproduce: install kodi-rbp

To fix I need to: chmod 0777 /dev/tty0

mcurly, 2015-06-06:
@joaodriessen maybe you have already solved this but I am posting here the solution I have used myself:
change the file /etc/udev/rules.d/raspberrypi.rules from

SUBSYSTEM=="vchiq|input", MODE="0777"
KERNEL=="mouse*|mice|event*", MODE="0777"

SUBSYSTEM=="vchiq|input|tty", MODE="0777"
KERNEL=="mouse*|mice|event*|tty0", MODE="0777"

There is one important note: I am not definitely sure if this will not harm the security of the system so if anyone else can shed a light on this topic I would be delighted.

Cheers.

pejobo, 2015-08-20:
See also https://wiki.archlinux.org/index.php/Kodi#Raspberry_Pi where the udev rule 
```
   SUBSYSTEM=="tty", KERNEL=="tty0", GROUP="tty", MODE="0666"
```
is suggested.