archlinuxarm/PKGBUILDs #1191: community/nzbget: Only runnable as root

evils, 2015-05-23:
Package: community/nzbget
Architecture: armv7h
Device: Raspberry Pi 2

What it isn't doing: 
su nzbget -c "nzbget -c /etc/nzbget.conf -D" -> no output, no created process nothing
the same thing under root though, works! (Though the server mode -s, works)
Tried with several other options, no luck. Seems to be a problem that was created during compilation

What it should be doing:
create the daemon

Steps to reproduce:
run "nzbget -D" from any non root user

kmihelich, 2015-05-23:
So what is the compile problem, and what have you done to prove that?

evils, 2015-05-23:
It's a suspicion that it's a compile problem. 
I've installed the package on 3 different ARMv7H devices and tried to run the daemon under a non root user, same behaviour across all devices.

kmihelich, 2015-05-23:
Suspicions aren't bugs.  Use the forum for usage issues.

closed 2015-05-23 by kmihelich

evils, 2015-05-23:
It's not an usage issue if the bin is damaged... 
Fine, I'll wait till you guys see what I've been telling