archlinuxarm/PKGBUILDs #1192: core/linux-raspberrypi 3.18.13-2: Unable to initialise dma channel on mmc-bcm

evils, 2015-05-23:
Package: core/linux-raspberrypi 3.18.13-2
Architecture: armv7h
Device: Raspberry PI 2

What it isn't doing:
initialising the DMA channels on the mmc-bcm2835.
Error message: "mmc-bcm2835 3f300000.mmc: mmc-bcm2835: Unable to initialise DMA channels. Falling back to PIO"

What it should be doing:
initialise the DMA channels on the mmc-bcm2835 and not in PIO

Steps to reproduce:
Install the latest linux-raspberrypi package on a rpi2
Also see:
http://archlinuxarm.org/forum/viewtopic.php?f=60&t=8874

evils, 2015-05-23:
I suspect another bug in this package that's not connected to the topic of this issue, but I'll write it down here as it's a suspicion that I wasn't able to confirm yet.

When using the lirc_rpi module and attaching an IR sensor such as the TSOP 4838 to the GPIO the pins are not allocated as they should, ```cat /sys/kernel/debug/gpio``` stays blank for these ports and the IR sensor doesn't work. 
Two mentions are here, one from me, one from another guy. Both on RPI2's with the latest kernels:
http://archlinuxarm.org/forum/viewtopic.php?f=57&t=8851
http://forum.archphile.org/viewtopic.php?f=6&t=149

If that goes to another package (even though I suspect it has to do with the mmc-bcm2835 chip, please let me know and I'm more than happy to open another issue for this.

kmihelich, 2015-05-26:
Fixed in 3.18.14-1.

closed 2015-05-26 by kmihelich

evils, 2015-05-27:
@kmihelich 

Not really. Tested on 3.18.14-1

```
[    4.972507] mmc-bcm2835 3f300000.mmc: mmc-bcm2835: Unable to initialise DMA channels. Falling back to PIO
```
lirc also doesn't work.

moonman, 2015-05-27:
https://github.com/raspberrypi/linux/issues/985

rpi_lirc works for me on rpi. Issue may be related to your setup or may be
specific to RPI2. Either way it is not related to this bug report.

evils, 2015-05-27:
@moonman Thanks for pointing out https://github.com/raspberrypi/linux/issues/985.
I haven't seen that. 

Let's hope it's fixed soon, as it's definitely not fixed in 3.18.14-1