archlinuxarm/PKGBUILDs #1202: kodi-rbp: incorrect owner of /usr/share/polkit-1/rules.d

jaystrictor, 2015-06-13:
Package: kodi-rbp
Architecture: armv7
Device: Raspberry Pi 2

What it isn't doing: The system isn't rebooting when issuing
``systemctl reboot``, but instead:
```
[root@alarmpi ~]# systemctl reboot
Error getting authority: Error initializing authority: Error calling StartServiceByName for org.freedesktop.PolicyKit1: Timeout was reached (g-io-error-quark, 24)
Failed to start reboot.target: Connection timed out
[root@alarmpi ~]# systemctl -l status polkit
* polkit.service - Authorization Manager
   Loaded: loaded (/usr/lib/systemd/system/polkit.service; static; vendor preset: disabled)
   Active: inactive (dead)
     Docs: man:polkit(8)

Jun 13 18:54:39 alarmpi polkitd[1332]: Lost the name org.freedesktop.PolicyKit1 - exiting
Jun 13 18:54:39 alarmpi systemd[1]: Started Authorization Manager.
Jun 13 19:03:28 alarmpi systemd[1]: Starting Authorization Manager...
Jun 13 19:03:28 alarmpi polkitd[1350]: Started polkitd version 0.112
Jun 13 19:03:28 alarmpi polkitd[1350]: Loading rules from directory /etc/polkit-1/rules.d
Jun 13 19:03:28 alarmpi polkitd[1350]: Loading rules from directory /usr/share/polkit-1/rules.d
Jun 13 19:03:28 alarmpi polkitd[1350]: Error opening rules directory: Error opening directory '/usr/share/polkit-1/rules.d': Permission denied (g-file-error-quark, 2)
Jun 13 19:03:28 alarmpi polkitd[1350]: Finished loading, compiling and executing 1 rules
Jun 13 19:03:28 alarmpi polkitd[1350]: Lost the name org.freedesktop.PolicyKit1 - exiting
Jun 13 19:03:28 alarmpi systemd[1]: Started Authorization Manager.
[root@alarmpi ~]# ls -lad /usr/share/polkit-1/rules.d/
drwx------ 2 root root 4096 Apr 19 18:27 /usr/share/polkit-1/rules.d/
[root@alarmpi ~]#
```

What it should be doing: polkit should work normally.
I think that the owner of ``/usr/share/polkit-1/rules.d/`` should be ``polkitd``.

Steps to reproduce:
1. Install a minimal alarmpi system.
2. ``pacman -S kodi-rbp``
3. ``pacman -S polkit``
4. Try to actually use polkit, e.g. by doing ``systemctl reboot``

I think this occurs when ``kodi-rbp`` is installed **before** ``polkit`` or any other package that owns ``/usr/share/polkit-1/rules.d/``.
The owner of the directory ``/usr/share/polkit-1/rules.d/`` is erroneously set to ``root``, although the correct owner would be ``polkitd``.

Furthermore let me add that IMHO this is probably also a bug in the package ``polkit``, as package ``polkit`` should set the owner correctly when being installed after any other package.
So this should be fixed both in any ``kodi-*`` package as well as in ``polkit``.

jaystrictor, 2015-06-13:
I reported this to the archlinux bug tracker as well (for package ``polkit``).
See https://bugs.archlinux.org/task/45318

kmihelich, 2015-06-13:
This is not a bug in the kodi packages.  polkit should be setting ownership of its directories in its install file and not in the install() section of the PKGBUILD to account for this directory pre-existing.

closed 2015-06-13 by kmihelich

Scimmia22, 2015-06-14:
You're joking, right?

kmihelich, 2015-06-14:
No, I'm not.  It is not be the responsibility of packages other than polkit to ensure that polkit's directories have the ownership rights needed by polkit.

Scimmia22, 2015-06-14:
It is the responsibility of the package to assign correct ownership of any directories it includes.

kmihelich, 2015-06-14:
Then by your terms, polkit should not be assigning any ownership over /usr/share/polkit-1/rules.d/.  I assume this will be corrected in your PKGBUILD shortly, as it is not that package's responsibility.

Scimmia22, 2015-06-14:
polkit includes that directory, of course it should be assigning ownership? polkit uses a static UID for a reason, so that other packages can include files and directories owned by that user. It most certainly is the packager's responsibility.

kmihelich, 2015-06-14:
And what I'm saying is that the ownership should be assigned in polkit.install to account for the fact that the directory may already exist on the system, given that polkit requires the directory to have certain access rights in order to properly function.

Assigning correct ownership in install files is not without precedent.  It is done in many packages, and prevents this exact issue from occurring.

Scimmia22, 2015-06-14:
Anyway, I closed the Arch report. This doesn't affect Arch as every package that includes this dir either has a hard dep on polkit and/or has the correct ownership; the one exception will be fixed. I suggest you  do the same

I'm out.

kmihelich, 2015-06-14:
Yeah, I'm not going to add one more thing to remember because a maintainer refuses to move two lines from one file to another.  You can do things the hard way, I'll fix it here.

kmihelich, 2015-06-14:
My bad, I assumed the maintainer (or any maintainer, for that matter) was making this decision.  This is Arch packaging policy being made by the bug wrangler.

This issue is fixed in polkit 0.112-2.1 (f87f82a)

Scimmia22, 2015-06-14:
No, but long standing practices which don't cause problems in Arch don't qualify as a bug.

kmihelich, 2015-06-14:
Then consider it a feature request and let the *package maintainer* decide if it is an invalid request or not.

jaystrictor, 2015-06-14:
Thank you both for trying your best in pushing your ego aside and handling this matter in a grown-up way.

kmihelich, 2015-06-14:
Really?  You want to start fires with personal attacks?  Don't bother filing further issues until you yourself grow up.