archlinuxarm/PKGBUILDs #1216: docker:bug in 1.7.0 on ARM

troyfontaine, 2015-07-10:
Package: docker 1.7.0
Architecture: ARMv7h
Device: Raspberry Pi 2

What it isn't doing: Cannot build container images - "socket operation on non-socket" error - known issue in 1.7.0 of Docker as per https://github.com/docker/docker/issues/14184

What it should be doing: Container should build without error.  Is fixed in 1.7.1

Steps to reproduce: Attempt to build a container using Docker 1.7.0-you will receive this error.

WarheadsSE, 2015-07-10:
Ah, wasn't aware it had been released, but yes I know.

WarheadsSE, 2015-07-10:
once it is released, I will update it. It has not yet been tagged and released.

pavel, 2015-07-12:
@WarheadsSE Iht would be great if you've could put docker's 1.7.1 latest rc, for example, as docker-rc aur package.

GerardRibas, 2015-07-13:
@pavel did you solve the issue? I'm having the same problem as you

WarheadsSE, 2015-07-13:
@GerardRibas the above issue @ docker/docker is sorted in 1.7.1-rc1 as that states. I will not however package that for the distribution.

pavel, 2015-07-13:
@WarheadsSE Got the 1.7.1-rc3 thing build, but when doing `docker run hello-world` or building a container I get: `[8] System error: exec format error`.

@GerardRibas If you're interested here's the modified PKGBUILD for the 1.7.1-rc3:

WarheadsSE, 2015-07-13:
Exec format errors are because you are attempting to run x86-64 containers

troyfontaine, 2015-07-13:
@WarheadsSE Sorry, I submitted the ticket when I was already on auto-pilot at the keyboard and missed that the release wasn't ready.  I've downgraded Docker in the interim until they release the new version.  Thanks for the update!

pavel, 2015-07-14:
@WarheadsSE My bad. Sure I can not run x86-64 containers on ARM CPU. Docker works fine.

commit 3616b030 (2015-07-15): community/docker to 1.7.1-1

fixes #1216
closed 2015-07-15 by kmihelich