archlinuxarm/PKGBUILDs #1222: kodi-imx 15.0 crashes on start when reading from database

cellscape, 2015-07-31:
After upgrade to Isengard on Cubox i4x4 Kodi failed to start. After some investigation I found that it crashes if there are any movies in database. The issue is reliably triggered with empty profile after adding few movies. However exactly the same profile/library works fine on x86_64 desktop, I believe the issue is specific to platform.

I've rebuilt kodi with debug symbols to get proper backtrace:
```
#0  0x0040fe20 in pair<Field&&, 0u> (__tuple2=<synthetic pointer>, __tuple1=..., this=<optimized out>) at /usr/include/c++/5.2.0/tuple:1172
#1  pair<Field&&> (__second=..., __first=..., this=<optimized out>) at /usr/include/c++/5.2.0/tuple:1161
#2  construct<std::pair<const Field, CVariant>, const std::piecewise_construct_t&, std::tuple<Field&&>, std::tuple<> > (__p=<optimized out>, this=<optimized out>)
    at /usr/include/c++/5.2.0/ext/new_allocator.h:120
#3  _S_construct<std::pair<const Field, CVariant>, const std::piecewise_construct_t&, std::tuple<Field&&>, std::tuple<> > (__p=<optimized out>, __a=...)
    at /usr/include/c++/5.2.0/bits/alloc_traits.h:256
#4  construct<std::pair<const Field, CVariant>, const std::piecewise_construct_t&, std::tuple<Field&&>, std::tuple<> > (__p=<optimized out>, __a=...)
    at /usr/include/c++/5.2.0/bits/alloc_traits.h:402
#5  _M_construct_node<const std::piecewise_construct_t&, std::tuple<Field&&>, std::tuple<> > (__node=0x7086dab0, this=0x64659160) at /usr/include/c++/5.2.0/bits/stl_tree.h:529
#6  _M_create_node<const std::piecewise_construct_t&, std::tuple<Field&&>, std::tuple<> > (this=0x64659160) at /usr/include/c++/5.2.0/bits/stl_tree.h:546
#7  std::_Rb_tree<Field, std::pair<Field const, CVariant>, std::_Select1st<std::pair<Field const, CVariant> >, std::less<Field>, std::allocator<std::pair<Field const, CVariant> > >::_M_emplace_hint_unique<std::piecewise_construct_t const&, std::tuple<Field&&>, std::tuple<> >(std::_Rb_tree_const_iterator<std::pair<Field const, CVariant> >, std::piecewise_construct_t const&, std::tuple<Field&&>&&, std::tuple<>&&) (this=0x64659160, __pos=...) at /usr/include/c++/5.2.0/bits/stl_tree.h:2170
#8  0x00c18b5c in operator[] (__k=<unknown type in /usr/lib/kodi/kodi.bin, CU 0x584a99c, DIE 0x58718dd>, this=0x0) at /usr/include/c++/5.2.0/bits/stl_map.h:502
#9  DatabaseUtils::GetDatabaseResults (mediaType=<error reading variable: Cannot access memory at address 0x20>, fields=std::vector of length 0, capacity 0, 
    dataset=std::unique_ptr<dbiplus::Dataset> containing 0x7082bc18, results=std::vector of length 0, capacity 3) at DatabaseUtils.cpp:369
#10 0x00c6b20c in SortUtils::SortFromDataset (sortDescription=..., mediaType="movie", dataset=std::unique_ptr<dbiplus::Dataset> containing 0x7082bc18, 
    results=std::vector of length 0, capacity 3) at SortUtils.cpp:735
#11 0x003c5a28 in CVideoDatabase::GetMoviesByWhere (this=this@entry=0x646596d4, strBaseDir="videodb://recentlyaddedmovies/", filter=..., items=..., sortDescription=...)
    at VideoDatabase.cpp:6215
#12 0x003c69bc in CVideoDatabase::GetRecentlyAddedMoviesNav (this=this@entry=0x646596d4, strBaseDir="videodb://recentlyaddedmovies/", items=..., limit=limit@entry=0)
    at VideoDatabase.cpp:6527
#13 0x015f6488 in XFILE::VIDEODATABASEDIRECTORY::CDirectoryNodeRecentlyAddedMovies::GetContent (this=0x70829d60, items=...) at DirectoryNodeRecentlyAddedMovies.cpp:38
#14 0x015f26b0 in XFILE::VIDEODATABASEDIRECTORY::CDirectoryNode::GetChilds (this=this@entry=0x70813330, items=...) at DirectoryNode.cpp:264
#15 0x00721be0 in XFILE::CVideoDatabaseDirectory::GetDirectory (this=<optimized out>, url=..., items=...) at VideoDatabaseDirectory.cpp:55
#16 0x006b1140 in XFILE::CDirectory::GetDirectory (url=..., items=..., hints=..., allowThreads=allowThreads@entry=false) at Directory.cpp:204
#17 0x006b33d0 in XFILE::CDirectory::GetDirectory (strPath="videodb://recentlyaddedmovies/", items=..., hints=..., allowThreads=allowThreads@entry=false) at Directory.cpp:131
#18 0x006b3448 in XFILE::CDirectory::GetDirectory (strPath="videodb://recentlyaddedmovies/", items=..., strMask="", flags=flags@entry=0, allowThreads=allowThreads@entry=false)
    at Directory.cpp:125
#19 0x007eb610 in CDirectoryJob::DoWork (this=0x26792c0) at DirectoryProvider.cpp:63
#20 0x00c2a1c0 in CJobWorker::Process (this=0x31a1c10) at JobManager.cpp:68
#21 0x0175a5b0 in CThread::Action (this=0x31a1c10) at Thread.cpp:221
#22 0x0175a84c in CThread::staticThread (data=0x31a1c10) at Thread.cpp:131
#23 0x76ab7e0c in start_thread () from /usr/lib/libpthread.so.0
#24 0x75ba3d60 in ?? () from /usr/lib/libc.so.6
```

Valgrind output:
```
==9352== Thread 16 JobWorker:
==9352== Use of uninitialised value of size 4
==9352==    at 0x40FE20: pair<Field&&, 0u> (tuple:1172)
==9352==    by 0x40FE20: pair<Field&&> (tuple:1161)
==9352==    by 0x40FE20: construct<std::pair<const Field, CVariant>, const std::piecewise_construct_t&, std::tuple<Field&&>, std::tuple<> > (new_allocator.h:120)
==9352==    by 0x40FE20: _S_construct<std::pair<const Field, CVariant>, const std::piecewise_construct_t&, std::tuple<Field&&>, std::tuple<> > (alloc_traits.h:256)
==9352==    by 0x40FE20: construct<std::pair<const Field, CVariant>, const std::piecewise_construct_t&, std::tuple<Field&&>, std::tuple<> > (alloc_traits.h:402)
==9352==    by 0x40FE20: _M_construct_node<const std::piecewise_construct_t&, std::tuple<Field&&>, std::tuple<> > (stl_tree.h:529)
==9352==    by 0x40FE20: _M_create_node<const std::piecewise_construct_t&, std::tuple<Field&&>, std::tuple<> > (stl_tree.h:546)
==9352==    by 0x40FE20: std::_Rb_tree_iterator<std::pair<Field const, CVariant> > std::_Rb_tree<Field, std::pair<Field const, CVariant>, std::_Select1st<std::pair<Field const, CVariant> >, std::less<Field>, std::allocator<std::pair<Field const, CVariant> > >::_M_emplace_hint_unique<std::piecewise_construct_t const&, std::tuple<Field&&>, std::tuple<> >(std::_Rb_tree_const_iterator<std::pair<Field const, CVariant> >, std::piecewise_construct_t const&, std::tuple<Field&&>&&, std::tuple<>&&) (stl_tree.h:2170)
==9352==    by 0xC18B5B: operator[] (stl_map.h:502)
==9352==    by 0xC18B5B: DatabaseUtils::GetDatabaseResults(std::string const&, std::vector<Field, std::allocator<Field> > const&, std::unique_ptr<dbiplus::Dataset, std::default_delete<dbiplus::Dataset> > const&, std::vector<std::map<Field, CVariant, std::less<Field>, std::allocator<std::pair<Field const, CVariant> > >, std::allocator<std::map<Field, CVariant, std::less<Field>, std::allocator<std::pair<Field const, CVariant> > > > >&) (DatabaseUtils.cpp:369)
==9352==    by 0xC6B20B: SortUtils::SortFromDataset(SortDescription const&, std::string const&, std::unique_ptr<dbiplus::Dataset, std::default_delete<dbiplus::Dataset> > const&, std::vector<std::map<Field, CVariant, std::less<Field>, std::allocator<std::pair<Field const, CVariant> > >, std::allocator<std::map<Field, CVariant, std::less<Field>, std::allocator<std::pair<Field const, CVariant> > > > >&) (SortUtils.cpp:735)
==9352==    by 0x3C5A27: CVideoDatabase::GetMoviesByWhere(std::string const&, CDatabase::Filter const&, CFileItemList&, SortDescription const&) (VideoDatabase.cpp:6215)
==9352==    by 0x3C69BB: CVideoDatabase::GetRecentlyAddedMoviesNav(std::string const&, CFileItemList&, unsigned int) (VideoDatabase.cpp:6527)
==9352==    by 0x15F6487: XFILE::VIDEODATABASEDIRECTORY::CDirectoryNodeRecentlyAddedMovies::GetContent(CFileItemList&) const (DirectoryNodeRecentlyAddedMovies.cpp:38)
==9352==    by 0x15F26AF: XFILE::VIDEODATABASEDIRECTORY::CDirectoryNode::GetChilds(CFileItemList&) (DirectoryNode.cpp:264)
==9352==    by 0x721BDF: XFILE::CVideoDatabaseDirectory::GetDirectory(CURL const&, CFileItemList&) (VideoDatabaseDirectory.cpp:55)
==9352==    by 0x6B113F: XFILE::CDirectory::GetDirectory(CURL const&, CFileItemList&, XFILE::CDirectory::CHints const&, bool) (Directory.cpp:204)
==9352==    by 0x6B33CF: XFILE::CDirectory::GetDirectory(std::string const&, CFileItemList&, XFILE::CDirectory::CHints const&, bool) (Directory.cpp:131)
==9352==    by 0x6B3447: XFILE::CDirectory::GetDirectory(std::string const&, CFileItemList&, std::string const&, int, bool) (Directory.cpp:125)
==9352==    by 0x7EB60F: CDirectoryJob::DoWork() (DirectoryProvider.cpp:63)
==9352==  Uninitialised value was created by a stack allocation
==9352==    at 0xC18224: DatabaseUtils::GetDatabaseResults(std::string const&, std::vector<Field, std::allocator<Field> > const&, std::unique_ptr<dbiplus::Dataset, std::default_delete<dbiplus::Dataset> > const&, std::vector [... cut ...]
==9352==    by 0x6B113F: XFILE::CDirectory::GetDirectory(CURL const&, CFileItemList&, XFILE::CDirectory::CHints const&, bool) (Directory.cpp:204)
==9352==    by 0x6B33CF: XFILE::CDirectory::GetDirectory(std::string const&, CFileItemList&, XFILE::CDirectory::CHints const&, bool) (Directory.cpp:131)
==9352==    by 0x6B3447: XFILE::CDirectory::GetDirectory(std::string const&, CFileItemList&, std::string const&, int, bool) (Directory.cpp:125)
```

Looking at code there are no uninitialized variables and segfault happens inside of stack-allocated std::map. Could there be a problem with compiler or standard library? Any ideas?

closed 2015-07-31 by kmihelich

beam, 2015-08-06:
It's current package releated, i have my own package builded with old libcec-imx6 v2 library and Isengard working with my MySQL library without problem.

Now i tried clean archlinux install on new SD card and kodi-imx package from repository and Kodi crashing after i set advancedsettings.xml to connet my library. Don't know if it working with local db.

cellscape, 2015-08-09:
It seems that recompiling with gcc-5.1 allows to run kodi-imx-15.0 just fine. Looks like recent update to gcc-5.2 is to blame.

I don't understand why the issue is closed is "invalid". Anyone trying to use kodi-imx with library on recent arch linux arm is stuck with this silly crash and boot loop. Basically package is unusable in current state. I understand that problem is probably in gcc/kodi and should be fixed upstream but for now why not to recompile it with gcc-5.1 and put working package into repo? At least it will be usable.

moonman, 2015-08-10:
First, when submitting issues you are required to read and follow the rules: https://github.com/archlinuxarm/PKGBUILDs/blob/master/CONTRIBUTING.md
Secondly there is no way going back to gcc 5.1 so we need to figure out what needs to be done to make kodi compile with gcc 5.2.

rkorn86, 2015-08-12:
@cellscape did you find a workaround? Im hitting the same bug on my utilite (imx6 platform) running a custom 3.10.70 kernel.
@moonman could you please post any link, where you or any other dev work on "figure out what needs to be done to make kodi compile with gcc 5.2"?

beam, 2015-08-18:
It's not only imx6 problem, same on rbp2. Kodi 15.1 & gcc 5.2 doesn't work.

SjonHortensius, 2015-08-20:
Fwiw, this is also reported [here](http://forum.kodi.tv/showthread.php?tid=235883). Can an ArchlinuxArm developer comment on wsnipex's comment?

>  Just note that gcc-5.2 requires your COMPLETE system to be rebuilt against the new ABI or really esoteric crashes can and will happen.

Also, I'm not sure if kodi-rpb-15.1 should really be in the repo as it currently won't work? Can't it be build with gcc-5.1?

moonman, 2015-08-20:
I just built kodi 15.1 yesterday and tried it on my cubox i. Added some
files to the db and it works. No crashes. Can't reproduce this.