archlinuxarm/PKGBUILDs #1225: systemd: systemd-networkd segmentation fault when using DHCP

osaukko, 2015-08-04:
Package: systemd
Architecture: armv7l
Device: Raspberry Pi 2

What it isn't doing:
    systemd-networkd does not get settings from DHCP. Well actually it does get
    settings but then crashes with SIGSEGV, Segmentation fault. Running
    /usr/lib/systemd/systemd-networkd with gdb shows that application does get
    valid settings from out DHCP server, creates new thread and then crashes.
    
    Last two lines from GDB:
    Program received signal SIGSEGV, Segmentation fault.
    0x76e38c6e in strlen () from /usr/lib/libc.so.6

What it should be doing:
    Apply network settings from DHCP server.

Steps to reproduce:
    Install all updates or create new installation using:
    http://archlinuxarm.org/os/ArchLinuxARM-rpi-2-latest.tar.gz
    
    This do happen now with clean installation on every boot. Before updates my
    RPi2 did work just fine.

Additional notes:
    Works fine when static IP address is configured to /etc/systemd/network/eth0.network file.

kmihelich, 2015-08-04:
Fixed with systemd 224-1.

closed 2015-08-04 by kmihelich

osaukko, 2015-08-05:
Thank you for fast fix. Tested on my RPI and it works fine again :)