archlinuxarm/PKGBUILDs #1227: chromium 44.0.2403.130-1: segmentation fault upon startup

iegubkin, 2015-08-09:
Package: chromium 44.0.2403.130-1
Architecture: armv7
Device: Raspberry Pi 2

What it isn't doing: Fails to start with error:
[24055:24055:0809/035431:ERROR:sandbox_linux.cc(345)] InitializeSandbox() called with multiple threads in process gpu-process
Segmentation fault (core dumped)

Don't know if this is a fault of packaging or something else. chromium 44.0.2403.107-1 started successfully

The error is affecting others:
http://archlinuxarm.org/forum/viewtopic.php?f=60&t=9109&p=47963&hilit=chromium#p47963

What it should be doing:

Steps to reproduce:

iegubkin, 2015-08-09:
Reverted to chromium 44.0.2403.107-1 and it starts successfully (no segmentation fault) but still gives the error:

```[26005:26005:0809/181041:ERROR:sandbox_linux.cc(345)] InitializeSandbox() called with multiple threads in process gpu-process
```

qjcg, 2015-08-09:
Package: 44.0.2403.130-1
Architecture: armv7 3.10.82-3-ARCH
Device: Odroid-XU3-lite

I've got the same issue.

iegubkin, 2015-08-14:
It would be interesting to know if the package maintainer is experiencing this problem. New version, same error:  44.0.2403.155-1

RaumZeit, 2015-08-24:
This seems to be related to some other package. Out of curiosity, I just compiled 43.0.2357.134 using the PKGBUILD of an earlier commit (448f514) and still experience segfaulting chromium. Maybe it's a compiler issue? I compiled with gcc version 5.2 from the alarm repo.

gripped, 2015-08-31:
Also out of curiosity I made compiled 44.0.2403.157-1 with gcc 5.1.0-5 and the corresponding binutils, glibc and libtool. 
The resulting chromium ran fine. With gcc 5.2 as the compiler it segfaults at start-up.

qjcg, 2015-09-02:
Just upgraded to version 45.0.2454.85-1 and I confirm that running chromium still results in a segfault every time.

Like @iegubkin, wondering if maintainer @foutrelis and/or others are running versions of the package from 44.0.2403.130-1 - current successfully.

kmihelich, 2015-09-02:
The x86 package maintainer does not have anything to do with our package, and "me too" comments are not useful. The issue with this package has and is being looked into, and will be fixed when it is fixed. Submit a PR if you have something to contribute.