archlinuxarm/PKGBUILDs #1244: kodi-{rbp,imx}: Segmentation fault upon loading (bigger) database file

Freso, 2015-09-03:
Package: kodi-rbp, kodi-imx
Architecture: armv6, armv7
Device: Raspberry Pi 2, Raspberry Pi model B, Cubox i4x4

What it isn't doing:
Booting without segfaulting; see output pasted in #1222 and at http://trac.kodi.tv/ticket/16206

What it should be doing:
Boot without segfaulting.

Steps to reproduce:

1. Have an already populated database (not sure what the criteria needed for this to trigger is though, seems it can happen both with local and MariaDB databases)
2. Upgrade to Kodi 15.1 built with GCC 5.2.
3. Try to start Kodi

The issue was previously closed in #1222, but it's also been closed upstream at http://trac.kodi.tv/ticket/16206 so I'm making a new issue, following the proper steps. (There's also a [forum thread mention](http://forum.solid-run.com/kodi-xbmc-player-on-cubox-i-and-hummingboard-f7/archlinuxarm-official-kodi-t636-s450.html) for anyone interested in that.) There is a sample db for reproduction step 1 on the Kodi ticket.

Signupnow, 2015-09-09:
Maybe this is the same issue as [#1242](https://github.com/archlinuxarm/PKGBUILDs/issues/1242)?

Freso, 2015-09-10:
@Signupnow #1242 appears to be MariaDB specific. I already mentioned in the original comment that this has happened for users both using MariaDB but also the default (SQLite) databases.

Signupnow, 2015-09-10:
@Freso OK, so my issue is a particular case of your issue.

paschty, 2015-09-10:
Is there a Workarround ? 
I dont have a old pgk of kodi (fresh setup) and the current version is not usable.
It would be very nice if someone could provide the latest working package until the problem is fixed.

Freso, 2015-09-10:
I've uploaded the last working version of kodi-rbp for armv7h (that is, the version for Raspberry Pi 2) at http://www27.zippyshare.com/v/iUHuEmxl/file.html - I don't have archived versions around for other versions/boards.

Signupnow, 2015-09-10:
@paschty Latest good one for armv6h (Raspberry Pi 1 A/B) here: http://www48.zippyshare.com/v/UNGTgtvF/file.html

paschty, 2015-09-11:
Thanks @Freso @Signupnow  (Tested the armv7h Version and it works)

moonman, 2015-09-12:
Check the forum maybe. There is a working build for 15.1 and explanation of what happened.

Freso, 2015-09-16:
kodi-rbp 15.1-2 seems to have this issue resolved, even if it hasn't hit this repository yet(?).

asbachb, 2015-09-16:
I think kodi-rbo 15.1-2 is just 15.1-1 compiled with gcc 5.1 (see http://archlinuxarm.org/forum/viewtopic.php?f=60&t=9090&start=30#p48581)

closed 2015-09-16 by kmihelich