archlinuxarm/PKGBUILDs #1258: kodi-rbp 15.1-1: this version does not load up to the GUI on a RPi2

graysky2, 2015-09-13:
Package: kodi-rbp 15.1-1
Architecture: armv7l
Device: RPi2

What it isn't doing: I cannot get the GUI to successfully load (all I see is 15.1 splash screen followed by the blue background). No menus appear.  My setup uses nfs shares to provide media

What it should be doing: Loading and providing a functional package. Perhaps kodi-rbp 15.1-1 needs to be rebuilt against libnfs?

Steps to reproduce: 
Configure some nfs shares for kodi to use and add the content to the library.
Run this version of kodi (systemd service) on an up-to-date RPi2

Solution: The problem is that you are serving up the package that was built with GCC5.2 which has known issues[1].  Using moonman's builds of with GCC5.1 fixes the problem[2],

Recommend the repo version gets built against GCC5.1 to prevent others from falling into this trap.

1. http://forum.kodi.tv/showthread.php?tid=238558&pid=2105159#pid2105159
2.  http://archlinuxarm.org/forum/viewtopic.php?f=60&t=9090&start=10#p48253

graysky2, 2015-09-13:
Related:
#1257
#1256

kmihelich, 2015-09-13:
Do not open another issue on this tracker. Ever.

closed 2015-09-13 by kmihelich