archlinuxarm/PKGBUILDs #1264: 3rd attempt - lua51 armv7 links to libncursesw.so.5 while lua does not

git-marijus, 2015-09-21:
ldd /usr/bin/lua5.1
	linux-vdso.so.1 (0xbef88000)
	libm.so.6 => /usr/lib/libm.so.6 (0xb6ea7000)
	libdl.so.2 => /usr/lib/libdl.so.2 (0xb6e92000)
	libreadline.so.6 => /usr/lib/libreadline.so.6 (0xb6e49000)
	libhistory.so.6 => /usr/lib/libhistory.so.6 (0xb6e31000)
	libncursesw.so.5 => /usr/lib/libncursesw.so.5 (0xb6dcd000)
	libc.so.6 => /usr/lib/libc.so.6 (0xb6c8e000)
	/lib/ld-linux.so.3 (0xb6f24000)

ldd /usr/bin/lua
	linux-vdso.so.1 (0xbea9c000)
	libm.so.6 => /usr/lib/libm.so.6 (0xb6eef000)
	libdl.so.2 => /usr/lib/libdl.so.2 (0xb6eda000)
	libreadline.so.6 => /usr/lib/libreadline.so.6 (0xb6e91000)
	libc.so.6 => /usr/lib/libc.so.6 (0xb6d52000)
	/lib/ld-linux-armhf.so.3 (0xb6f6c000)
	libgcc_s.so.1 => /usr/lib/libgcc_s.so.1 (0xb6d25000)
	libncursesw.so.6 => /usr/lib/libncursesw.so.6 (0xb6cc1000)

Expected would be that /usr/bin/lua5.1 also links to libncursesw.so.6 like lua version 5.3 does.

WarheadsSE, 2015-09-21:
https://www.archlinux.org/packages/extra/i686/lua51/

This package has not been touched upstream Arch since 2013.

git-marijus, 2015-09-21:
ldd /usr/bin/lua5.1 in upstream arch gives me this:
	linux-vdso.so.1 (0x00007ffd67dc7000)
	libm.so.6 => /usr/lib/libm.so.6 (0x00007f412481c000)
	libdl.so.2 => /usr/lib/libdl.so.2 (0x00007f4124618000)
	libreadline.so.6 => /usr/lib/libreadline.so.6 (0x00007f41243ce000)
	libc.so.6 => /usr/lib/libc.so.6 (0x00007f412402a000)
	/lib64/ld-linux-x86-64.so.2 (0x00007f4124b1a000)
	libncursesw.so.6 => /usr/lib/libncursesw.so.6 (0x00007f4123dbd000)

so it is fine in upstream x86_64 architecture

WarheadsSE, 2015-09-21:
Well then someone upstream has been with their toys wrong. You're fully up to date?

git-marijus, 2015-09-21:
up to date arch and archlinuxarm yes...

git-marijus, 2015-09-21:
Will rebuilding lua51 resolve this issue?

commit 038577c5 (2015-09-22): extra/lua51 to 5.1.5-4.1

closes #1264
closed 2015-09-22 by kmihelich