archlinuxarm/PKGBUILDs #1267: xf86-video-armsoc-odroid: No longer compiles

gripped, 2015-09-26:
Package: xf86-video-armsoc-odroid
Architecture: armv7
Device: odroid XU3

What it isn't doing: Compiling successfully

What it should be doing: Compiling successfully

Steps to reproduce: Run makepkg -As

```
make[2]: Entering directory '/home/odroid/test-repo/xf86-video-armsoc-odroid/src/xf86-video-armsoc-odroid/src'
  CC       drmmode_display.lo
drmmode_display.c: In function ‘drmmode_handle_uevents’:
drmmode_display.c:1760:14: error: storage size of ‘s’ isn’t known
  struct stat s;
              ^
drmmode_display.c:1773:6: error: implicit declaration of function ‘fstat’ [-Werror=implicit-function-declaration]
  if (fstat(pARMSOC->drmFD, &s)) {
      ^
drmmode_display.c:1773:2: error: nested extern declaration of ‘fstat’ [-Werror=nested-externs]
  if (fstat(pARMSOC->drmFD, &s)) {
  ^
drmmode_display.c:1760:14: error: unused variable ‘s’ [-Werror=unused-variable]
  struct stat s;
              ^
cc1: all warnings being treated as errors
Makefile:502: recipe for target 'drmmode_display.lo' failed
make[2]: *** [drmmode_display.lo] Error 1
make[2]: Leaving directory '/home/odroid/test-repo/xf86-video-armsoc-odroid/src/xf86-video-armsoc-odroid/src'
Makefile:440: recipe for target 'all-recursive' failed
make[1]: *** [all-recursive] Error 1
make[1]: Leaving directory '/home/odroid/test-repo/xf86-video-armsoc-odroid/src/xf86-video-armsoc-odroid'
Makefile:372: recipe for target 'all' failed
make: *** [all] Error 2
==> ERROR: A failure occurred in build().
    Aborting...
```
Full output
http://pastebin.com/17dHMr4P

Some members at the odroid forum have been working on a possible fix for the armsoc bug on the XU3/4 whereby switching from a full screen window back to a normal window often results in a black screen with only the mouse cursor showing.

While trying to apply the patch I realised that the package no longer builds?

kmihelich, 2015-09-26:
We are not the developers of this software.

closed 2015-09-26 by kmihelich

gripped, 2015-09-26:
You are not the developers of much of the software but you do sometimes create and apply patches to get things working. This did use to compile so something in Arch must have changed.

If I can figure it out I will create a pull request.
My guess is it could be related to the C standard being used but that is just a guess.
My hope was that one you developers , with far more expertise than me, would take one look and just know :)