archlinuxarm/PKGBUILDs #1275: linux-imx6-cubox-dt: Routing policy rule list corrupted

tempestnano, 2015-10-11:
Package: linux-imx6-cubox-dt
Architecture: armv7h
Device: Cubox-i4pro

What it isn't doing: 
Version 3.14.54-1 of this package causes strangeness with my routing policies.Only ip rule 0 is used by the system as a result. I would suggest a revert to 3.14.53 until this is corrected upstream.

Version 3.14.54-1
```
#ip rule list
0:      from all lookup local
0:      from all lookup local
0:      from all lookup local
0:      from all lookup local
0:      from all lookup local
0:      from all lookup local
0:      from all lookup local
0:      from all lookup local
0:      from all lookup local
0:      from all lookup local
```
This output will continue forever if I let it.

What it should be doing:

Version 3.14.53-1 (Nothing altered save the downgrade of linux-imx6-cubox-dt)
```
# ip rule list
0:      from all lookup local
32765:  from all fwmark 0x3 lookup transmission
32766:  from all lookup main
32767:  from all lookup default

```

Steps to reproduce:
I believe all one needs to do is run "ip rule list" using the new kernel.

moonman, 2015-10-11:
Will wait for next release, if it isn't fixed will need to dig for an offending commit

moonman, 2015-10-11:
https://git.kernel.org/cgit/linux/kernel/git/stable/linux-stable.git/commit/?id=427f994587dde5a0064e85614a631c075481c4ba is what causes this bahaviour.

Fixed in https://github.com/archlinuxarm/PKGBUILDs/commit/157a45de948a900713b6c90663ff2f90ce47c22b

closed 2015-10-11 by moonman