archlinuxarm/PKGBUILDs #1284: raspberrypi-firmware: compile from source instead of using raspbian binaries

zxalexis, 2015-10-18:
Package: raspberrypi-firmware
Architecture: armv6h, armv7h
Device: raspberry pi, raspberry pi 2
Broadcom recently published source for userland - compiling /opt/vc/* binaries and libs from source will build them against our kernel and libraries instead of raspbian-prebuilt binaries.

The code is ugly and not clean at all - but i've managed to build it using minor patches. At least, now it works as expected on RPi2 with alarm.
replaces: raspberrypi-firmware, raspberrypi-firmware-examples, raspberrypi-firmware-tools
Source tarball: http://www.filedropper.com/raspberrypi-userland-git-20150918-1srctar

kmihelich, 2015-10-18:
Where is the upstream source located?

zxalexis, 2015-10-18:
https://github.com/raspberrypi/userland

kmihelich, 2015-10-18:
Will be taken under consideration.

closed 2015-10-18 by kmihelich