archlinuxarm/PKGBUILDs #1285: mongodb: segmentation fault launching versions above mongodb-3.0.4

jaduff, 2015-10-22:
Package: mongodb-3.0.6-2-armv7h.pkg.tar.xz, mongodb-3.0.7-2-armv7h.pkg.tar.xz
Architecture: Armv7
Device: Raspberry Pi 2

What it isn't doing: Starting. systemctl start mongodb results in segmentation fault. Log below.

What it should be doing: Starting and running without error.

Steps to reproduce: Update from package mongodb-3.0.4-1-armv7h.pkg.tar.xz, systemctl start mongodb. Restarting in between does not improve.

***********************************************
"systemctl status mongodb" result below:
$ systemctl status mongodb -l
● mongodb.service - High-performance, schema-free document-oriented database
   Loaded: loaded (/usr/lib/systemd/system/mongodb.service; disabled; vendor preset: disabled)
   Active: failed (Result: exit-code) since Thu 2015-10-22 20:01:10 AWST; 3s ago
  Process: 311 ExecStart=/usr/bin/mongod --quiet --config /etc/mongodb.conf (code=exited, status=14)
 Main PID: 311 (code=exited, status=14)

Oct 22 20:01:09 raspberryarch systemd[1]: Started High-performance, schema-free document-oriented database.
Oct 22 20:01:10 raspberryarch mongod[311]: 2015-10-22T20:01:10.212+0800 F -        Invalid access at address: 0
Oct 22 20:01:10 raspberryarch mongod[311]: 2015-10-22T20:01:10.243+0800 F -        Got signal: 11 (Segmentation fault).
Oct 22 20:01:10 raspberryarch mongod[311]: 0xb89558 0xb88c18 0xb8902c 0x763a8a30
Oct 22 20:01:10 raspberryarch mongod[311]: ----- BEGIN BACKTRACE -----
Oct 22 20:01:10 raspberryarch mongod[311]: {"backtrace":[{"b":"10000","o":"B79558"},{"b":"10000","o":"B78C18"},{"b":"10000","o":"B7902C"},{"b":"7637C000","o":"2CA30"}]}
Oct 22 20:01:10 raspberryarch systemd[1]: mongodb.service: Main process exited, code=exited, status=14/n/a
Oct 22 20:01:10 raspberryarch systemd[1]: mongodb.service: Unit entered failed state.
Oct 22 20:01:10 raspberryarch systemd[1]: mongodb.service: Failed with result 'exit-code'.
jaduff@raspberryarch ~ $ systemctl status mongodb -l
● mongodb.service - High-performance, schema-free document-oriented database
   Loaded: loaded (/usr/lib/systemd/system/mongodb.service; disabled; vendor preset: disabled)
   Active: failed (Result: exit-code) since Thu 2015-10-22 20:01:10 AWST; 2min 47s ago
  Process: 311 ExecStart=/usr/bin/mongod --quiet --config /etc/mongodb.conf (code=exited, status=14)
 Main PID: 311 (code=exited, status=14)

Oct 22 20:01:09 raspberryarch systemd[1]: Started High-performance, schema-free document-oriented database.
Oct 22 20:01:10 raspberryarch mongod[311]: 2015-10-22T20:01:10.212+0800 F -        Invalid access at address: 0
Oct 22 20:01:10 raspberryarch mongod[311]: 2015-10-22T20:01:10.243+0800 F -        Got signal: 11 (Segmentation fault).
Oct 22 20:01:10 raspberryarch mongod[311]: 0xb89558 0xb88c18 0xb8902c 0x763a8a30
Oct 22 20:01:10 raspberryarch mongod[311]: ----- BEGIN BACKTRACE -----
Oct 22 20:01:10 raspberryarch mongod[311]: {"backtrace":[{"b":"10000","o":"B79558"},{"b":"10000","o":"B78C18"},{"b":"10000","o":"B7902C"},{"b":"7637C000","o":"2CA30"}]}
Oct 22 20:01:10 raspberryarch systemd[1]: mongodb.service: Main process exited, code=exited, status=14/n/a
Oct 22 20:01:10 raspberryarch systemd[1]: mongodb.service: Unit entered failed state.
Oct 22 20:01:10 raspberryarch systemd[1]: mongodb.service: Failed with result 'exit-code'.

TPXP, 2015-10-29:
Also see the related forum post http://archlinuxarm.org/forum/viewtopic.php?f=15&t=9374 and issue #1282

drmonty, 2015-10-30:
Same issue, here, but with odroid-xu3

nodeleaf, 2015-11-10:
Same issue here with package mongodb-3.0.7-2 on RPi 2B:

**mongod start**
`2015-11-10T16:21:51.140+0000 F -        Invalid access at address: 0`
`2015-11-10T16:21:51.148+0000 F -        Got signal: 11 (Segmentation fault).`

 `0xb89558 0xb88c18 0xb8902c 0x763a8a30`
`----- BEGIN BACKTRACE -----`
`{"backtrace":[{"b":"10000","o":"B79558"},{"b":"10000","o":"B78C18"},{"b":"10000","o":"B7902C"},{"b":"7637C000","o":"2CA30"}]}`
` mongod(_ZN5mongo15printStackTraceERSo+0x2C) [0xb89558]`
` mongod(+0xB78C18) [0xb88c18]`
` mongod(+0xB7902C) [0xb8902c]`
` libc.so.6(__default_rt_sa_restorer+0x0) [0x763a8a30]`
`-----  END BACKTRACE  -----`

monkeygroover, 2015-11-25:
+1, same issue on Raspberry pi (1 and 2)