archlinuxarm/PKGBUILDs #1288: linux-armv7: not generating initramfs

glebchek, 2015-10-25:
Package: linux-armv7
Architecture: armv7
Device: cubieboard 2 (it is not relevant)

What it isn't doing:
It is not generating initramfs image after install/upgrade
So if you encrypted root and rely on initramfs image to boot correctly, you should generate it manually and even then after next kernel upgrade boot will be broken

What it should be doing:
linux-armv7.install should call mkinitcpio during install and upgrade, like it is in linux-armv8 package

Steps to reproduce:
install package

WarheadsSE, 2015-10-30:
And what of systems not making use of initramfs?

glebchek, 2015-11-04:
I think there are 2 ways:
1. Initramfs will be generated on all systems, even if they don't use it. It is not such a cpu- and time- consuming task
2. Initramfs will be generated only on upgrade and only if initramfs image is already present on the system. So those who make use of initramfs will have to run mkinitcpio only once and those who doesn't wont be affected at all. But such behaviour should be documented somewhere

In #1289 I changed names of preset file and initramfs image to linux-armv7.preset and initramfs-linux-armv7.img accordingly so it doent collide with custom kernels but bootloader script still searches for initramfs-linux.img file by default.

moonman, 2015-11-04:
There shouldn't be a fallback image like on x86 as there is no choice presented on boot up to even try to load it. I also think it should have the name uboot expects which is initramfs-linux.img (just like all uboots expect zImage) but WarheadsSE has a point too so I don't know what's better.

I have been using initrd forever just for fsck on bootup and personally I don't think there is any harm in adding initrd and any devices that have no support for it just won't load it. What do @WarheadsSE and @kmihelich think?

kmihelich, 2015-11-05:
This is not a bug, and not what this issue tracker is for, per the *first* line of the contributing guidelines.

closed 2015-11-05 by kmihelich