archlinuxarm/PKGBUILDs #1295: core/linux-armv8 naming

ctmarinas, 2015-11-10:
Hi,

Please rename the core/linux-armv8 package and the PKGBUILD references to either "linux-aarch64" or "linux-arm64" (the latter would match the actual "arch/arm64" kernel structure. The reason is that while ARMv8 is the latest version of the architecture, it supports both AArch32 and AArch64. The ARMv8-A implementations have AArch64, however, running a 32-bit kernel should not be excluded (either directly or as a guest). The ARMv8-R profile only has the AArch32 mode at EL1 (with full MMU support), so not possible to run an AArch64 Linux kernel even though it is called ARMv8. Quite likely that at some point we'll see ARMv8 patches for the AArch32 kernel (this has been done for some time for the AArch32 gcc). Please change the name before it becomes too confusing.

Thanks.

P.S. And thanks for the great work on enabling Arch on AArch64 (running it on a Juno board).

kmihelich, 2015-11-10:
Sure, this can be done. It seems to me that linux-aarch64 would best describe its target.

closed 2015-11-11 by kmihelich

ctmarinas, 2015-11-11:
BTW, there is core/linux-armv8-rc as well ;).

kmihelich, 2015-11-11:
That'll get changed in 5-6 days with 4.4.

commit b07daf1b (2015-11-11): rename core/linux-armv8 to core/linux-aarch64

fixes #1295