archlinuxarm/PKGBUILDs #1298: firefox-42.0-2: Package firefox-42.0-2 broken; Firefox won't start after upgrade from firefox-42.0-1

wamserma, 2015-11-13:
Package: firefox-42.0-2
Architecture: armhf
Device: HP Chromebook 11 (Spring)

What it isn't doing: Firefox won't start (can't find libxul)

Pakete (1) firefox-42.0-2

~~~~~
[markus@archbook pkg]$  sudo pacman -Sy firefox
Gesamtgröße der zu installierenden Pakete:  91,75 MiB
Größendifferenz der Aktualisierung:        0,00 MiB

:: Installation fortsetzen? [J/n] J
(1/1) Prüfe Schlüssel im Schlüsselring             [######################] 100%
(1/1) Überprüfe Paket-Integrität                   [######################] 100%
(1/1) Lade Paket-Dateien                           [######################] 100%
(1/1) Prüfe auf Dateikonflikte                     [######################] 100%
(1/1) Überprüfe verfügbaren Festplattenspeicher    [######################] 100%
(1/1) Aktualisiere firefox                         [######################] 100%
[markus@archbook pkg]$ firefox
XPCOMGlueLoad error for file /usr/lib/firefox/libxul.so:
libvpx.so.3: cannot open shared object file: No such file or directory
Couldn't load XPCOM.
~~~~~

What it should be doing: find libxul

Steps to reproduce: sudo pacman -Sy firefox 
Reproducible after downgrade from cache and upgrade with fresh download. firefox-42.0-1 works fine.

wamserma, 2015-11-13:
Fix: Dependencies seem to be incorrect. firefox-42.0-2 should depend on libvpx-1.5.0-2

kmihelich, 2015-11-13:
Partial upgrades are unsupported.

closed 2015-11-13 by kmihelich