archlinuxarm/PKGBUILDs #1321: GCC and man-pages conflict

vendion, 2015-12-14:
When trying to do a system update I ran into two file conflicts between gcc and man-pages, it seems both packages provide _/usr/share/man/man3/random.3.gs_ and _/usr/share/man/man3/regex.3.gz_.  It seems that Upstream Arch Linux also hit this problem a [few days ago](https://bugs.archlinux.org/task/47279) in `[testing]` and seemed to have resolved it but I can't seem to find the change.

WarheadsSE, 2015-12-14:
man-pages 4.02-4 had this fix. 4.03-1 is in the repos, straight from ABS. Try installing the updated man-pages before the gcc update.

madalu, 2015-12-14:
The conflict occurs with 4.03-1. Installing man pages first only changes the warning to "exists in filesystem" when trying to install gcc:

error: failed to commit transaction (conflicting files)
gcc: /usr/share/man/man3/random.3.gz exists in filesystem
gcc: /usr/share/man/man3/regex.3.gz exists in filesystem
Errors occurred, no packages were upgraded.

WarheadsSE, 2015-12-14:
Oddly enough, despite that task in the Arch Linux bug tracker: 

$ tar tf /var/cache/pacman/pkg/man-pages-4.03-1-any.pkg.tar.xz | grep random
usr/share/man/man4/random.4.gz
usr/share/man/man4/urandom.4.gz
usr/share/man/man3/random.3p.gz
usr/share/man/man3/random.3.gz
usr/share/man/man3/random_r.3.gz

madalu, 2015-12-14:
man-pages 4.03-1 has this error. I get the followin error if I instal man-pages first.

error: failed to commit transaction (conflicting files)
gcc: /usr/share/man/man3/random.3.gz exists in filesystem
gcc: /usr/share/man/man3/regex.3.gz exists in filesystem
Errors occurred, no packages were upgraded.

madalu, 2015-12-14:
The change is in gcc. 

$ pacman -Qlp gcc-5.2.0-2-arm.pkg.tar.xz | grep -E "/man3/(random|regex).3.gz"
[no results]

$ pacman -Qlp gcc-5.3.0-2-arm.pkg.tar.xz | grep -E "/man3/(random|regex).3.gz"
gcc /usr/share/man/man3/random.3.gz
gcc /usr/share/man/man3/regex.3.gz

WarheadsSE, 2015-12-14:
[this](https://projects.archlinux.org/svntogit/packages.git/commit/trunk?h=packages/man-pages&id=522b8836a3f735b58dfe2eeb4bc6f26a448eb379) is claiming it was fixed instead in gcc. However, I can't find any matching changes in GCC's recent commits.

closed 2015-12-15 by kmihelich